prusa3d/Prusa-Firmware-Buddy #3740: [BUG] Mk3.5 info show wrong bootloader version

Lino77, 2024-02-10:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MK3.5] show incorekt bootloader see pic 
[image]

arekm, 2024-02-12:
Note - when using "English" it shows correct bootloader version (2.3.4).

bkerler, 2024-02-13:
@Lino77 Can you change to english language, does it show the right bootloader version ? And if you change back to german, does it still show the wrong version ?

Lino77, 2024-02-13:
English shoes the right  bootloader back to german Shows thr wrong

bkerler, 2024-02-29:
Ok, will dig into the issue

Tomcus, 2024-06-10:
This issue seems to be fixed in 6.0.X, can you confirm that @Lino77 and then close the issue?

closed 2024-06-10 by Lino77