prusa3d/Prusa-Firmware-Buddy #3741: [BUG] Setting safety timer with M86 while nozzle or bed are heated immediately triggers timer

PinkPandaKatie, 2024-02-10:
**Printer type** - MK4 but likely affects all

**Printer firmware version** - 5.1.2

**Original or Custom firmware** - Original

**Optional upgrades** - None

**USB drive or USB/Octoprint** - USB

**Describe the bug**
Sending `M86 S2400` to change the safety timer (or any value other than 0) while heating is active will immediately trigger the timer and show "Heating disabled" instead of resetting it.

Looking at the code in [/src/marlin_stubs/feature/safety_timer.cpp](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/master/src/marlin_stubs/feature/safety_timer.cpp#L29), it looks like it should be setting `last_reset` to `millis()`, but instead it sets it to the passed interval `ms`. This means that if the printer has been turned on for more than twice the specified interval, it will immediately trigger.

**How to reproduce**
  Heat nozzle with `M104 S215` then send `M86 S2400`

**Expected behavior**
  Safety timer should reset and only trigger after the specified interval

**G-code**
  N/A

**Crash dump file**
  N/A

bkerler, 2024-02-13:
Yeah, I think you are right. I wrote a PR #3752 for the issue.

CZDanol, 2024-02-14:
Internal ticket: BFW-5077

CZDanol, 2024-02-14:
Fix merged. Should appear in the next full release :)

closed 2024-02-14 by CZDanol