prusa3d/Prusa-Firmware-Buddy #3742: [BUG] MK3.5 - X Axis check fails

balcanusebi, 2024-02-10:
Printer: MK3.5 Original Firmware 5.2.0
The X axis test fails. Please see attached video.

https://github.com/prusa3d/Prusa-Firmware-Buddy/assets/22098405/cbd749ef-2f5a-491d-a078-3242b014d55b

bkerler, 2024-02-18:
@balcanusebi The video is rather small. Can you do a higher quality video so that the x values can be seen on the display while the self-test is running ?

balcanusebi, 2024-02-18:
> @balcanusebi The video is rather small. Can you do a higher quality video so that the x values can be seen on the display while the self-test is running ?

Sure, I've attached a new video.
Thanks for having a look at it!

https://github.com/prusa3d/Prusa-Firmware-Buddy/assets/22098405/1c6803c5-4b74-49ce-8ca6-d879ae7395ec

bkerler, 2024-02-18:
Ok, I think the problem might be that when the x-endstop is reached, the value is 250.67, but it should be 251 instead. Is there some kind of obstacle on the right size of your extruder or maybe the bearing that is too far on the right side ?

bkerler, 2024-02-18:
Note: I'm referring to my MK4 here, no idea if for the MK3.5 250.67 is ok, but I don't think so.

balcanusebi, 2024-02-19:
@bkerler there’s no obstacle on the X axis. I’ve even went that far and disassembled the entire X axis, replaced the bearings and made sure that everything is as it should be according to the instructions manual. 
The weird thing is that, with the MK3S+, the X axis checked without any issues. 
Would it be an idea to increase the “tolarence” on the X length? For example, everything that goes over the value of 250, to be considered as passed. 
Thanks for you help!

bkerler, 2024-02-19:
I need to review the firmware, but I think this is something Prusa needs to have a look.

Quantenradierer, 2024-03-18:
It looks like your heater fan screws are not fully in. Only the lower right fan screw seems to be correctly in.
Try to screw them more in - or remove them and test it without the screws at first.

The screws are propably hitting the y-axis on the left side when you test the x-axis, therefore the missing 3.3mm.

closed 2024-07-02 by github-actions[bot]