prusa3d/Prusa-Firmware-Buddy #3743: User Addition of Filament Types To The Printer

KlausKraemer, 2024-02-11:
I miss a possibilty to add new filament types to the printer. They don't have to be hard coded in the FW, they could be stored in a csv, xml or whatever easy editable file on the USB stick.

This could be connected with Prusa Slicer on editing according filament settings or simply manually editing that file.

postatemullerdotnl, 2024-02-11:
The issue with mismatching filament lists is that you are unable to start an remote print. (you have to be physically near the knob to say 'yes yes I know the filament types are not matching')

KlausKraemer, 2024-02-11:
This could also be solved via Prusa Connect asking for confirmation. In my case its a 98% probability, I am near the printer.

bkerler, 2024-02-11:
See related PR #3657. There are already several issues for this topic.

Prusa-Support, 2024-06-28:
This issue can actually be closed as a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1258.

Michele Moramarco
Prusa Research