prusa3d/Prusa-Firmware-Buddy #3744: [FEATURE REQUEST] per object thumbnails

beelsebob, 2024-02-11:
When you print a bunch of instances of an object, it can be hard to tell which object is which and which to cancel.  This could be vastly improved if the slicer included thumbnails for each object, that could then be displayed at the cancel object screen.

The thumbnails should be just like the main thumbnail, except that only on object should be orange.  The rest should be greyed out.

**Printer type** - mk4
**Optional upgrades** - I mean, it’s a frankenprinter, but do you care?

bkerler, 2024-02-13:
Actually that's a feature request that mainly has to happen in prusaslicer first. Can you please open up an issue there and refer to the issue here ?

CZDanol, 2024-12-13:
Hello, I've created a pitch for the product guys for this as an internal ticket BFW-6465