prusa3d/Prusa-Firmware-Buddy #3745: [BUG] [MK3.5] Board temperature missing - "Not initialized"

arekm, 2024-02-12:
**Printer type** - MK3.5
**Printer firmware version** - 5.2.0
**Original or Custom firmware** - Original

**Describe the bug**
Buddy board temperature is not shown.

**How to reproduce**
Go to "Sensor info" menu and notice there is "Not initialized" instead of board temperature.

[image]

**Expected behavior**
Board temperature shown, just like on MK4 firmware.

bkerler, 2024-02-13:
Indeed. Looks like the code for the board temp wasn't added for an unknown reason. Here's the PR #3749 that fixes this issue.

CZDanol, 2024-02-13:
Internal ticket number: BFW-5074

CZDanol, 2024-02-13:
Fix merged :) Should got into the next release.

magistersmax, 2024-03-03:
I updated to 5.2.1 today and the board temperature is still showing as Not Initialized.  

[image]
[image]

arekm, 2024-03-03:
Same here on 5.2.1.

bkerler, 2024-03-05:
People, just because it's merged doesn't mean it's already in the next release. Relax.

arekm, 2024-03-05:
Well, it was mentioned in 5.2.1 announcement as fixed.

bkerler, 2024-03-05:
Indeed, I just checked and although the bugfix is mentioned in the v5.2.1 release notes, the PR isn't included in the commits @JohnnyDeer @CZDanol.

CZDanol, 2024-03-05:
Yep, it was supposed to go there. Sorry, guys, looks like administrative error. I've notified the team about that.

bkerler, 2024-03-06:
Thanks @CZDanol

HonzaJaros, 2024-03-20:
Sorry for writing here, but this exactly is happening in MK4 fw 5.1.3.
[image]
[image]