prusa3d/Prusa-Firmware-Buddy #3746: [BUG] [MK3.5] Hotend fan speed - "N/A" despite self test passing

arekm, 2024-02-12:
**Printer type** - MK3.5
**Printer firmware version** - 5.2.0
**Original or Custom firmware** - Original

**Describe the bug**

  A clear and concise description of what the bug is.

**How to reproduce**
Go to "Sensor info" menu and notice there is "N/A" at "Hotend Fan".

**Expected behavior**
Just like Print Fan - "0 % / stopped" if it's stopped or the real speed if it is not.

Note that fan speed test passes just fine and speed of both fans is visible in rpms when fan self test is being made.

[image]
[image]

(heater test fails because I'm using Revo and 5.2.0 firmware is not ready for it)

NicholasRakita, 2024-02-14:
We're looking into this now.

Internal ticket: BFW-5076

arekm, 2024-02-14:
I see there is a fix

````
commit 8a2615d50bf39b684edaa6439d107ba954786932
Author: info@revskills.de <info@revskills.de>
Date:   Tue Feb 13 11:09:30 2024 +0100

    Fix missing board temperature display on MK3.5
````

michalrudolf, 2024-09-09:
Fixed and scheduled for upcoming MK3.5 release

danopernis, 2024-09-17:
Yes, this is actually the same issue as https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4088 which is fixed since v6.2.0-alpha1

closed 2024-09-17 by danopernis