prusa3d/Prusa-Firmware-Buddy #3747: [BUG] [MK3.5] Wrong machine type and extruder count reported on M115 (MACHINE_TYPE:Prusa-mini EXTRUDER_COUNT:5)

arekm, 2024-02-12:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3.5
**Printer firmware version** - 5.2.0
**Original or Custom firmware** - Original

**Describe the bug**

Printer reports on M115:

````
FIRMWARE_NAME:Prusa-Firmware-Buddy 5.2.0+13881 (Github) SOURCE_CODE_URL:https://github.com/prusa3d/Prusa-Firmware-Buddy PROTOCOL_VERSION:1.0 MACHINE_TYPE:Prusa-mini EXTRUDER_COUNT:5 UUID:...
````

**How to reproduce**

Issue ````M115```` command over a serial connection.

**Expected behavior**

MACHINE_TYPE should be probably MK3.5
EXTRUDER_COUNT should be 1

There was a similar bug on MK4 firmware but that one got fixed: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3153

CZDanol, 2024-02-14:
Internal ticket: BFW-5081

CZDanol, 2024-02-15:
Fix merged :)

closed 2024-02-15 by CZDanol