prusa3d/Prusa-Firmware-Buddy #3753: [ENHANCEMENT] Add the prime tower to list of cancelable objects

BaconFase, 2024-02-13:
Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - XL 5T

**Is your suggested improvement related to an existing problem? Please describe.**
Especially with tall prints, at some point you just get a few bad layers or blobs and the the tower comes falling down. Sometimes there are only certain layers that 'need' the tower and its just a time/filament waste the rest of the print. I'd like the ability to simply cancel the tower like cancelling any other object mid-print.

**Describe the expected functionality**
If using a prime tower, have it listed in the Cancel Objects menu on the LCD. If cancelling the tower during a print, it is skipped like any other cancelled object.

bkerler, 2024-02-15:
Not sure if that should be handled by the prusaslicer instead ?

BaconFase, 2024-02-16:
The main usage is when you have something like a 300 tall print and the purge tower falls over at z=150 which took a day to print.
The choices at the moment are:
1.  jank-rig a tower structure out of tape, hot glue, and old MMU purge towers so that the tower can keep printing thus keep the print alive
2. just keep printing the tower in midair and pray to whichever gods those filament strings don't strongly adhere to your main model
3. start over from scratch and lose all that time

Speaking from recent experience:
[image]

I'm simply suggesting to add the tower to the existing cancelable object list, so the printer can just completely ignore a failed tower and continue printing the main model(s). I don't understand how this could be handled at the slicer?

The other part (only needing the tower in certain layers) was just another hypothetical where the suggested change could be applied. Though, I'm not against having more control over the generated tower in PS.

KlausKraemer, 2024-02-17:
Another way would be a choice of a more solid tower in Prusa Slicer. The one and only tower is very flimsy, even though it often works. Compared to a big print gone wrong, having a little more waste should even be sustainable, before throwing the wrecked print away.

arekm, 2024-02-17:
https://github.com/prusa3d/PrusaSlicer/issues/11615

BaconFase, 2024-02-17:
Closed as copy of #11615

closed 2024-02-17 by BaconFase