prusa3d/Prusa-Firmware-Buddy #3754: [FEATURE REQUEST] Turn off nozzle immediately after load/unload

murk-sy, 2024-02-13:
**Printer type** - MINI + others
**Optional upgrades** - N/A

## **Is your feature request related to an existing problem? Please describe.**
Recently, the loading/unloading filament functionality has been changed to not heat the bed. However, I believe there is also no reason to continue heating the nozzle after a load/unload.

In my mind, especially considering the quantity of printers in the wild, this may actually also save a non negligible amount of power in the long term.

### After load:
Heating up the bed will (as far as I remember at least) **always** take longer than heating up the nozzle to ABL temperature. Therefore, keeping the nozzle heated for an extended period is not really worthwhile, in my mind. Some users may want to also clean the nozzle after a load, but I believe that is perfectly doable before the procedure is confirmed completed, or even after while it's cooling down.
Users wanting to preheat the printer while they slice/send gcode would likely want to preheat the bed most of all, which is easily accessible.

### After unload:
Since "change filament" is an option, the user is unlikely to need the hotend heated after an unload. If a service is planned, the user would either want it cooled to room temperature or at a specific, manually set temperature. The latter would waste a little bit of time at the worst, but the nozzle likely wouldn't even reach the idle temperature in the time the user could manually set it.

## **Describe the expected functionality**
After filament load/unload, instead of 170°C with a 30 minute timeout, the nozzle temperature is set to 0 immediately.

### If your workflow is different, please share your thoughts.
As a side note, I'm out of office, so I may be a bit off with how existing features work.

alexsporn, 2024-02-14:
This has been implemented in this open PR https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/3700

bkerler, 2024-02-15:
Duplicate of issue #3691, which has already been addressed in PR #3700.

murk-sy, 2024-02-23:
Closing as duplicate and adding my comment on that issue

closed 2024-02-23 by murk-sy