prusa3d/Prusa-Firmware-Buddy #3755: [FEATURE REQUEST] Filament runout on XL

Stippy19, 2024-02-14:
If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - XL

**Is your feature request related to an existing problem? Please describe.**
When the filament sensor on the side or toolhead recognize a filament runout, the printer will ask for new filament. It will then unload the filament, so that the user can grab it at the PTFE tube at the side sensor.

**Describe the expected functionality**
Instead of forcing an unload, ask the user if he wants to unload or if he manually wants to push in filament himself. I just manually push new filament trough the side sensor and the PTFE tube until it reaches the extruder itself to not wast that 50cm out of the PTFE tube by first unloading. Would be great to have that option.
Additionally, in this case, the printer could finish the filament off (part from side sensor to extruder) until the extruder sensor is also showing missing filament and then wait for the user to introduce new filament.

TechnoSwiss, 2024-02-15:
Yes, PLEASE!

Cap1a1n, 2024-02-17:
+1

Basically similar to [ENHANCEMENT] XL Filament sensor individual control of side and nextruder sensor #3257 https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3257

Woelpi, 2024-02-18:
This would actually be great. I'm also thinking about putting the filament completely elsewhere, to have less issues with TPU.

Stippy19, 2024-02-18:
Additional feature request would be: If the sensor is active and detects a filament running out, offer the possibility to disable this sensor. Currently you cannot do that and if the sensor repeatedly fails to recognize the filament (like when using TPU), then the print is just doomed. Would be great to have the possibility to switch the toolhead sensor off at that moment, too.

TechnoSwiss, 2024-02-19:
> Additional feature request would be: If the sensor is active and detects a filament running out, offer the possibility to disable this sensor. Currently you cannot do that and if the sensor [...]

Also a great idea, I've had this problem before as well and had to ditch a mostly complete print because the sensor didn't recognize the new filament.