prusa3d/Prusa-Firmware-Buddy #3757: [BUG] Inexplicable MINTEMP ERROR on MK4

Zappes, 2024-02-15:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK4

**Printer firmware version** - 5.1.3

**Original or Custom firmware** - Original

**Optional upgrades** - None

**USB drive or USB/Octoprint**
  PrusaConnect, i.e. probably USB

**Describe the bug**
  When printing with PETG (didn't happen with PLA so far), I occasionally get a beeping redscreen with the message "MINTEMP ERROR". 

The QR code leads to a KB article explaining that this error will trigger if the bed temperature falls below 5° at any point in time. This article told me to set the bed temp to 80°C and wiggle cables to see if the bed temperature fluctuates, which would point to a faulty thermistor. I performed this procedure and no fluctuation could be triggered by that.

The PrusaConnect and PrusaLink telemetry shows nothing suspicious. 
[image]
The crash occurred right at the moment where the nozzle temperature reached its peak, that is some seconds after the second layer started printing. This was done with a stock Prusament PETG profile, i.e. the bed temp for the first layer was set to 85°C and to 90°C for all subsequent layers.

My personal working hypothesis is that the check in the firmware might have issues as I never experienced this on 5.0.0 while it keeps occurring every third print or so on 5.1.3. In addition, this does suspiciously look as if that check noticed the temperature increase of the bed on the start of the second layer while ignoring that the target temperature had actually been altered. I could imagine that the reference value for the check is set at the beginning of the print and doesn't get updated after the first layer. My only test of this hypothesis, though, was reslicing the model where this occurred the last time with a bed temp of 85°C for all layers and printing that. The print is on the third layer while I write this, but I acknowledge that this doesn't mean a lot - in the past, it always sufficed to simply restart the failed print which would then go through flawlessly.

**How to reproduce**
I don't really know as it's been an intermittent problem so far. If above hypothesis is correct, though, it might be reproducible by choosing very different values for the first layer and other layer bed temps, perhaps something like 70°C and 90°C.

**Expected behavior**
The error should not occur unless there's an actual thermal anomaly.

**G-code**
GCode of the last print that caused the error: 
[Schalterrahmen Parametrisch 3x_PETG_.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/14295574/Schalterrahmen.Parametrisch.3x_PETG_.zip)

Zappes, 2024-02-15:
Sadly, the experiment with the lowered bed temperature failed, i.e. I got that same error, just later.
For Science, I will now try the same model with the enclosure doors open.

Zappes, 2024-02-15:
I am not a C++ developer and I surely don't know my way around the firmware, but when searching for the error i was shown, I found this part of the code that looks really fishy:

https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/860909b34e4cd7f116b4e803d1abd21dec2e145b/lib/Marlin/Marlin/src/module/temperature.cpp#L887

[image]

That looks as if the first string parameter might be a copy/paste error as the combination of MINTEMP and MAXTEMP looks really strange. Also, according to the code comment, this seems to be related to a sensor _above_ the heatbreak - that might match my problem description as everything here seems to point to something with the environmental temperature instead of the nozzle or bed.

Zappes, 2024-02-15:
@bkerler That's great, thanks! <3

No I only have to find out what the actual problem here is. The part that failed me miserably three times in a row with the enclosure doors shut is currently printing its last few layer with open doors - and I'd really like to know why... Given that the failures occurred at a mere 45°C, I am a bit afraid that I might face a forced non-printing time in the summer. :-/

bkerler, 2024-02-15:
It seems that the nozzle temp reached the max temperature, for whatever reason, leading to a max temp error, which was displayed as "min temp error" because the label was wrong of the window

Zappes, 2024-02-15:
Hmm. As my screenshots above show, this sounds at least slightly improbable - the nozzle temps were clearly OK during the failed prints.

Just asking: In the error message, there was also a QR code and some descriptive text. The QR code led to [this page](https://help.prusa3d.com/article/mintemp-error-and-mintemp-bed_2169), and the text clearly referred to a problem with the heat bed temperature as well. Would that be consistent with the issue you fixed? At first look, that code seemed to only print a wrong headline while the message body seems OK.

Zappes, 2024-02-15:
It rises because of the settings and the temperatures in the diagram above are exactly what they are supposed to be:

- First layer bed temp 85°C for the first layer and 90°C for the others in prints #1 and #2. For Print #3 I set both values to 85°C which gave me roughly 90 minutes before the crash instead of the 20 minutes I got for 85/90.
- First layer nozzle temp 240°C, all other layers 250°C. These are the values from the stock Prusament PETG profile, which seems fitting for the filament used. (Prusament PETG Galaxy Black)

The temperatures reported by the printer were at no time out of the valid ranges.

**Now the fun part.**

The three failed prints were done with the doors of my Original Prusa Enclosure closed. I restarted the print with the doors open, and that attempt looks extremely promising:

[image]

The enclosure never got to more then 45°C during the failed prints, as one can see in the temperature graph above. With open doors, I have quite constant 30°C in the enclosure.

Now, there doesn't seem to be a simple correlation like "above 40°, the print fails". As the diagrams above clearly show, the third print happily ran for more than an hour at an enclosure temperature significantly higher than it was at the previous failure times. It rather seems to be an issue that has something to do with the rate at which the environmental temperature _changes_.

All of these are just hypotheses, obviously. To be honest, I have no plausible ideas what could be wrong here, and some of my observations may be misleading. For example, I did encounter this issue a few times with firmware 5.1.2. After downgrading to 5.0.0, the problem went away. After updating to 5.1.3, the issue re-appeared, but until today I never saw it more than one time in a row... It's quite puzzling, really.

Zappes, 2024-02-19:
Just printed a really simple thing, enclosure doors were open. The simple thing contained a color change after the first layer, and while the extruder was happily purging, I saw my favourite message...
[image]
I will now revert to 5.0.0 again. I was told by a dev that this can't possibly help here, so I guess it's worth trying - even if it's just for eliminating one potential cause.

Zappes, 2024-02-19:
It certainly isn't enough evidence to pin this on the firmware version, yet - but I just completed two prints of the same object that failed in the above case. Both with the enclosure closed. If I should ever encounter the MINTEMP issue with 5.0.0, I will update this ticket - if no such update is to be seen, I have yet to encounter a situation where this actually happens with 5.0.0.

Zappes, 2024-02-20:
OK, it took some time to happen, but I did get the error with firmware 5.0.0 right now as well, thus eliminating the firmware as cause of the issue. I'll close this issue now and chat with Prusa support. :)

closed 2024-02-20 by Zappes

lezetuser, 2024-02-27:
@Zappes you issue has not been solved, the thing what @bkerler resolved it concerned the heatbreak thermal error not the bed MINTEMP error. There is only a coincidence that during the research you found it.

LibraryTechMF, 2024-05-21:
I just wanted to add to this, as I am having the same issue, also with PETG. In my case, it's a fairly long print of multiple parts. I have tried slicing it to both .bgcode and normal .gcode and encountered similar issues with both. It seems to happen sometime between 5 and 7 hours into the print. I am on the latest firmware 6.0.1.

bkerler, 2024-05-25:
Actually it was a hardware issue, and I highly assume that others having similar issues should also contact Prusa support.