prusa3d/Prusa-Firmware-Buddy #3758: [BUG] MK4 Heatbed self test fails but works fine due to wrong build plate.

TK0130, 2024-02-15:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MK4]

**Printer firmware version** - [5.0.1+12089]

**Original or Custom firmware** - [Original]

**USB drive or USB/Octoprint**
n/a

**Describe the bug**
Self test of bed fails unless smooth build plate is installed.

**How to reproduce**
Run self test with the textured, satin, or no build plate.

**Expected behavior**
Printer should request that user install the smooth build plate, or give the option to select a build plate, or no build plate.

  Do not share the file publicly, as the crash dump contains a raw snapshot of the printer's memory and may include unencrypted sensitive information.

TK0130, 2024-02-15:
Issue duplicated on latest firmware version:

Printer firmware version - [5.1.3+12089]

bkerler, 2024-02-15:
Actually I can't reproduce this issue, my MK4 always passes the selftest with satin build plate (and smooth of course).

TK0130, 2024-02-15:
Actually, on the latest firmware it passes with satin or smooth but fails with textured, aftermarket plate, and no build plate.

I feel like calibration should just have a note to remove the build plate and to do calibration with no build plate. (What if a user only has textured? Or only has an aftermarket build plate?)

Looks like it fails consistently at around 65C while warming up.

bkerler, 2024-02-18:
I just tested the textured plate with the full selftest on 5.1.3 on the MK4 and it passed without any issues from a cold status. It seems like your heatbed might be defective, please contact support. Clearly no bug here.

closed 2024-07-02 by github-actions[bot]