prusa3d/Prusa-Firmware-Buddy #3762: [BUG] <Enter comprehensive title>

Marinosbitter, 2024-02-16:
## The bug: ##
During power-on it gives an error:
Error! #17604
Firmware file missing in the USB flash!

Clicking on the rotary knob continues startup and printer functions as normal.

## General info ##
**Printer type** - Original Prusa XL (2 toolhead)
**Printer firmware version** - 5.1.2+13478
**SD card or USB/Octoprint** - SD Card

## Symptoms: ##
### When the error triggers: ###
* Power on with an empty USB gives the error
* Power on with an USB with gcode files gives the error
* Power on with an USB with firmware file gives the error
* Power on with USB inserted gives the error, but clicking the rotary knob continues the startup and from then on, the printer works  completely normal

### No error get triggered: ###
* Power on without USB inserted gives no error and starts normal
* Connecting a USB after startup works as normal
* Rest of the functionality of the printer works as normal, both after startup with the error as after startup without the error
* I can print form the USB
* I can flash the firmware using USB
* I can print from Prusa Connect
* I can upload files trough Prusa Connect
* I can flash the firmware using Prusa Connect
* Printer self tests are all OK

## Solutions tried: ##
I'm also working with Prusa Support trough email on this issue, but so far we haven't reached a solution. Things we tried:
* Different USB (both USBs work with my Prusa MINI as normal)
* Tried original USB that came with the printer
* Tried a SanDisk 32GB USB3.0
* Formatting (quick) both USBs
* Formatting (complete) both USBs
* Flashing to older firmware
* Flashing to same firmware
* Flashing to new firmware trough Prusa Connect
* Reconnecting the grey ribbon cable behind the LCD
* Reconnecting the orange ribbon cable behind the LCD

closed 2024-06-11 by Marinosbitter

riversedge, 2024-09-07:
So was there a fix?   Mine doing this too (5T) and trying to work through the problem.   Similar set of steps, including replacing the sandwich board but so far no fix.

danopernis, 2024-09-17:
@riversedge which firmware are you running? There was an update to the bootloader in recent v6.1.3 firmware which should fix some of those strange issues.

riversedge, 2024-09-17:
6.1.3 did fix it.