prusa3d/Prusa-Firmware-Buddy #3764: [QUESTION] [MK3.5] [Revo] Switch hotend selftest bit, so printer will think that self-test passed - how?

arekm, 2024-02-17:
"E3D REVO hotend
As of now, there might be issues with running the REVO hotend produced by the company E3D on the MK3.5. While testing this configuration, we discovered that due to the strict checks built into the firmware, the initial self-test will fail.

It is possible to start the print, however, we recommend waiting for the upcoming release, which will extend the support for 3rd party hotends."

https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.2.0

and my MK3.5 printer behaves exactly as described - prints well, just requires me to go to it and press a button *every time* I start a print job (and I'm starting prints from computer on a different floor).

So question:
**Is there a way to switch bit in memory that stores information about hot end self test to a value that it will think that test passed? If yes - how? **

That would make using 5.2.0 firmware with Revo much more pleasant while waiting for future firmware versions.

(8bit firmware had such feature via special gcode but no idea how that looks in 32bit fw)

mmcglumphy, 2024-02-18:
+1. I, too, would like to disable this check, if possible.

bkerler, 2024-02-18:
Actually there are three checks:
https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/b3ee47aa573f4d9b4af82d7a6963f43ab37ccc08/src/common/selftest/selftest_MK3.5.cpp#L381-L383

Which one do you mean ?

I'm not aware that there is any gcode to write eeprom addresses manually, so I assume a functionality needs to be added for that. But I guess it would be much better to add Revo support to the mk3.5 instead.
If you can give a more detailed example where the self-test fails, maybe a PR can be coded to support the MK3.5 with Revo hotend.

bkerler, 2024-02-18:
I see that an internal ticket BFW-5076 already exists for the issue (heater check fails), so Prusa is already working on it.

arekm, 2024-02-19:
hotend test (it's first half of "Heater Test")

Yes, they are working on it and this question is about interim workaround.

8bit fw had D3/D5 commands - https://help.prusa3d.com/article/prusa-firmware-specific-g-code-commands_112173#d-codes but if 32bit doesn't have something similar then probably there is no way to do workaround

beelsebob, 2024-02-23:
+1, but can we expand this to manually "passing" any part of the calibration.  My bear mk4 fails the y axis calibration (and gives exactly 0 information about what is wrong), but prints just fine.  It's super frustrationg to have to go and push a "get on with it" button every time I print.

grimbeaver, 2024-02-24:
I think the issue is with more than just the REVO.  I have a completely stock hotend except I replaced the original heater block with a E3D Copper Heater Block.  Not a huge change but I can not get the self-test for the hot end to pass.

arekm, 2024-03-19:
Selft test expects to get temperature between 180 and 230C after 60 seconds. Meeting that goal will make hotend self test pass.

On my side it was done by blowing into nozzle with air compressor with stopper running in that way, so after 60s it would end up in above temperature range. It worked!

https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/fb97a2d02bb0df53bc601695b66250f69cec7263/src/common/selftest/selftest_MK3.5.cpp#L106C1-L132C1

closed 2024-03-19 by arekm

arekm, 2024-03-20:
And freshly released 5.2.2 passes test with Revo without any such "hacks".

https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.2.2

TheMedGeek, 2024-03-20:
Still having the issue with a Slice Engineering Mosquito hotend :(

arekm, 2024-03-20:
Start a timer and tell us what temperature is shown for hotend after 60s (if it ever reaches 60s). Assuming hotend set to Revo in hardware settings.

TheMedGeek, 2024-03-20:
Thanks for the reply. It’s set to Revo - gets to 136 when the fail happens (60s)

arekm, 2024-03-20:
It heats too slowly to the expectations of the firmware. 

Don't set it to Revo (as it moves entire range up by 40C). Set it as "with sock" - that moves range down by 25C.

Help heating the nozzle with soldering iron (set to 200-300C) in a way, so it will end up between 155-205C range after 60 seconds. (Simply stop helping with iron if it heats too fast and would overshoot or bump iron temperature if too slow).

keyboard-magpie, 2024-04-02:
My 3.5 with revo 60w heater is consistently failing the hotend test, despite trying: parts fan blasting fully, heating up faster with soldering iron, all three possible options. Should we open a new issue?