prusa3d/Prusa-Firmware-Buddy #3771: [ENHANCEMENT] Display numbers in a user-friendly form

dhlavaty, 2024-02-19:
Printer: Prusa MK4
FW: v5.1.3

Would it be possible to display the numbers in a user-friendly form?

[image]

thomasloven, 2024-02-20:
That's approximately 1 % of the size of the visible universe. I don't think the way the number is displayed is your main problem...

dhlavaty, 2024-02-20:
The observable universe is a sphere with a diameter of about 93 billion light-years. So...

Can we agree that `1.0e+21 km` is not a user-friendly form?

thomasloven, 2024-02-20:
Don't mean to derail this topic, but I think 1.0e+21 km is rather user friendly. It's standard exponential notation for very large numbers and in my opinion better than `0.001 lyr` or `1000000000000000000000000 m` which would be two options to display roughly the same number.

Seriously though, something has gone horribly wrong in your printers estimation of the ammount of distance it has traveled - and expecting the user inteface to cater for that is pointless.

closed 2024-07-02 by github-actions[bot]