prusa3d/Prusa-Firmware-Buddy #3773: [BUG] Printer randomly freezes when printing, to then resume

FAB1150, 2024-02-19:
**Printer type** - MK4

**Printer firmware version** - 5.1.2+13478

**Original or Custom firmware** - Original

**Optional upgrades** - None

**Using an USB drive** - yes
 
**Describe the bug**
  As you can see in the video, the printer sometimes randomly freezes in the middle of a print. I originally thought it was the USB drive, but I noticed that the whole printer UI freezes (as you can see in the video). After 10-20 seconds, the printer just resumes as if nothing happened.

I have seen issue #3512 , but the person mentions something similar happening every few layers, so I think it might be a different problem. Here it's a rare problem, happens maybe 1-2 times every 6h of printing. I also attached some photos of artifacts that appear where the nozzle stops and plastic oozes out.

**How to reproduce**
  Just print any file, and it sometimes happens. Doesn't seem to happen in specific parts of the prints or fast movements, just random

**Expected behavior**
  Printer prints the part without freezing

**G-code**
  It happens with any .bgcode file. I am not sure about .gcode files, I think I noticed it with them too, but I'm not so sure. I'll try to reproduce with non-binary gcode too, and update this issue if it does.

**Video**
  https://youtu.be/p_dZydq1Atg
  Photos: https://photos.app.goo.gl/ZqNM5nhy89MRCu477

mix579, 2024-02-19:
Same here. Happens occasionally, impossible to reliably reproduce as the same file may print fine, then next time it may pause.

BaconFase, 2024-02-26:
Other than #3512, there is also https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3773

But I just wanted to add to the count of people encountering this issue, except on an XL-5T (5.1.2+13478, Prusa provided USB, no other connected hardware ), not an MK4. 

First time I encountered/noticed this pausing-issue was a print a few days ago that paused every several layers and blobbed a little out each time, but did finish. The next print plate of the same model finished without any noticed pausing nor blobs. My current print was hit by a pause once in the past 19hrs (https://youtu.be/VFVKZJInzoI), no blob. I've no clue what's up.

bkerler, 2024-02-29:
Double of #3512. Please don't reopen issues for the same problem.

FAB1150, 2024-03-01:
> Double of #3512. Please don't reopen issues for the same problem.

Is it though? I opened this issue specifically because I'm not sure if it's the same problem that causes the issue, as I mentioned.

bkerler, 2024-03-01:
There are already several issues, probably caused by the same reason. I assume it is related to reading from the usb stick vs. queueing, but didn't yet have time to look into.

FAB1150, 2024-03-02:
> I assume it is related to reading from the usb stick vs. queueing

I'm not sure about that (at least in my case) as the whole printer interface freezes. Unless the whole firmware is waiting for the USB to respond, it would be an unusual way to do things though

closed 2024-07-02 by github-actions[bot]