prusa3d/Prusa-Firmware-Buddy #3774: [BUG] Displayed Set Temperature Incorrect After Filament Load

magistersmax, 2024-02-20:
**Printer type** - MK3.5

**Printer firmware version** - 5.2.0

**Original or Custom firmware** - Original

**Optional upgrades** - N/A

**USB drive or USB/Octoprint** - USB

**Describe the bug**
After loading filament and confirming that the filament load is complete, the extruder temperature is lowered to 170C.  The displayed set temperature doe snot update to 170C, and stays on whatever temperature the filament was loaded at.  In my case the displayed set temperature corrected itself somehow, possibly when navigating to another screen?

**How to reproduce**
Enable extruder temperature display at the bottom of the screen.  Load a filament and confirm that the filament was loaded.  Watch the indicated current and set temperatures:  the current temperature should start falling to 170C, the set temperature will remain on the temperature used to load the filament.

**Expected behavior**
The indicated set temperature should update to 170C following filament load.

mix579, 2024-02-20:
That's intended behavior. Your "set" temperature is what you've set it to, 170 is the "parking" temperature to avoid oozing from the nozzle. So it shouldn't change the display for the set temperatur because 170 is not what you set it to.

magistersmax, 2024-02-20:
That makes sense, I’ll have to play with it some when I get home.  It did eventually change the displayed set temperature to 170 though, which is inconsistent with your description. 

Let me play with it when I get home to better determine what causes the set temperature to to to 170.

bkerler, 2024-02-20:
The nozzle_preheat_temp is 170, which isn't the nozzle_temp (which is higher). After unload, the last set temperature will be used. There is a PR #3700 to change that behaviour to optional cool down the nozzle after unload. But in general, this is expected behaviour.

magistersmax, 2024-02-21:
Yep, I played with it and I think I'm just wrong here.  Closing.

closed 2024-02-21 by magistersmax