prusa3d/Prusa-Firmware-Buddy #3775: [BUG] M0 command does not display the message string

DoubleStrike, 2024-02-20:
**Printer type** - XL 5 Toolhead

**Printer firmware version** - 5.1.2+13478

**Original or Custom firmware** - Original

**Describe the bug**
  Sending an M0 unconditional stop command does not display the message associated with the gcode.  While this is minimally functional, the message may be important and should always be shown.

**How to reproduce**
  Add the following line to any gcode:
`M0 You should see this message`

**Expected behavior**
  The printer should pause and display the user message. This is the expectation for a functional Marlin firmware, as per this page: https://marlinfw.org/docs/gcode/M000-M001.html.

bkerler, 2024-02-20:
I wrote the PR #3779 for your issue.

Prusa-Support, 2024-04-16:
Good news! The Pull Request was merged and just implemented in [FW 6.0.0-RC3](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.0.0-RC3).

Michele Moramarco
Prusa Research

closed 2024-04-16 by Prusa-Support