prusa3d/Prusa-Firmware-Buddy #3776: issue with both filament sensors

werkman, 2024-02-20:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [XL 1T]

**Printer firmware version** - [5.1.2]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [non]

**USB drive or USB/Octoprint**
USB drive

**Describe the bug**
  I have problems with de Prusa XL after updating the firmware to 5.1.2. I have issues with both the filament sensor in the tool head and the filament runout sensor. both have problems with detecting the filament and the sensors regularly switches between on and off and it is (almost) impossible to load filament with filament sensors switched on
For now the sensor disabled to print 

**How to reproduce**
  Switch sensor on and the trouble begins again

**Expected behavior**
  At the old firmware both sensors works without any problem. I expect they works better with the new firmware

**G-code**
  No need for G-code the trouble begins before I can print

**Crash dump file**
  No crash dump file available

**Video**
  On the video the unstable sensor activity on home screen while inserting the filament. Also at the end immediately unloading the filament due to sensor failure

https://github.com/prusa3d/Prusa-Firmware-Buddy/assets/7398003/a449cc55-7c3b-4b67-9e75-6fcec7d79e35

werkman, 2024-02-21:
> Did you contact Prusa Support ? Looks like a hw issue, not like a fw bug.
>
>
> Vlissingen K.v.K.: 41114014
> Disclaimer: hz.nl/disclaimer<http://hz.nl/disclaimer>

Hi Bjoern,

No I didn't because te issue starts after the update. Before the printer works without any problem. Never have had any problem with the filament sensors.
I'v made a picture from the sensor data. I'v no idea what's normal in this case.
The only problem can be, I came from an old firmware from around June 2023
Dit I miss an update between?

Jorien.

werkman, 2024-03-12:
after all the runout sensor wasnt working correctly.
The ball inside the sensor could not pop out far enough.
corrected it with a small screwdriver.

closed 2024-03-12 by werkman

bkerler, 2024-03-12:
Thanks for letting us know :)