prusa3d/Prusa-Firmware-Buddy #3777: [BUG] Menu Scrolling on XL is slow and laggy on 5.1.2

WhiteDragonD, 2024-02-20:
XL firmware 5.1.2

Updated from version 4.7.2, same USB stick, nothing changed.
When scrolling to select print, I have 17 folders with a few files in each.  

4.72 the scrolling was smooth, no lag.
5.1.2 the scrolling has issues, when you get to a 'new' page there's about a 2-5 second lag then if you keep scrolling up/down, the printer will crash.

**Expected behavior**
It should scroll as it did before without lagging.

Mryck, 2024-02-25:
Hello, I think it's the same issue as #3407

CZDanol, 2024-03-29:
Internal ticket BFW-4765, fix should be out in the upcoming 6.0 release. Please let me know if it fixes the issue :)

Closing the ticket for now.

closed 2024-03-29 by CZDanol