prusa3d/Prusa-Firmware-Buddy #3782: [ENHANCEMENT] Allow an unload/load/change for all 5 toolheads

bsikes1228, 2024-02-22:
**Printer type** - [XL]
**Optional upgrades** - [5 toolheads]

**Is your suggested improvement related to an existing problem? Please describe.**
  faster filament changes would be quite helpful

**Describe the expected functionality**
 one could choose to unload, load, change multiple toolheads at once to speed filament changes

murk-sy, 2024-03-14:
This is already a feature

https://help.prusa3d.com/article/lcd-menu-xl-multi-tool_500099

Filament > Change Filament in All Tools
Allows you to change filament in multiple toolheads at once (not necessarily *all* of them)

Before I used it for the first time I was also under the impression it would change every single filament, despite wanting to change only 2-3. Perhaps renaming it to "Change filament in multiple tools" would be clearer.

bsikes1228, 2024-03-14:
Thanks Murk for your message. Yes, I should have been more specific and
requested that the existing "change all" be tweaked to allow literally all
5 toolheads to be changed at once. Perhaps that's too much power drain at
once, but at least have the next in line in the change series heating when
you get to the load filament step of the first. I've never been able to get
it to change more than one at a time without the toolhead crashing, have
you? If so, please tell me how you get that to work...
On my machine, if I change to change more than one toolhead at a time, it
crashes after the first one is changed. I have to choose one, change it, go
back into the screen, select the next change, etc.
Still faster than using the single change command but...
Best,
B

murk-sy, 2024-03-14:
I'm *fairly* sure it starts preheating all heads when changing all, or maybe with a delay.

But toolhead crashes means that something is misaligned, not a firmware issue. I've never had issues running multiple filament changes in a row.
You should probably run the dock position calibration: https://help.prusa3d.com/guide/5-first-run_482742#482884

bsikes1228, 2024-03-14:
Just did it a week ago but will give it another go to make sure
Thnx

closed 2024-07-02 by github-actions[bot]