prusa3d/Prusa-Firmware-Buddy #3784: [BUG] Pause and Filament change command is ignored on Prusa Mini

dwbmb, 2024-02-23:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI+]

**Printer firmware version** - [5.1.2]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**USB drive or USB/Octoprint**
  USB flash drive

**Describe the bug**
  After generating bgcode in PrusaSlicer (v.2.7.1), loading it to the printer and printing, both g-code commands: pause print and filament change are being completely ignored. Printer just continues printing like there were no such a commands at all.

**How to reproduce**
  Generate binary gcode in PrusaSlicer 2.7.1. then try to print it on Prusa Mini+ FW: 5.1.2

**Expected behavior**
  I would expect it would pause on pause command and change filament on change filament command.

**G-code**
  Attached along with project .3mf file
[Large Rocket.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/14384507/Large.Rocket.zip)

**Crash dump file**
  N/A, printer not crashing

  Do not share the file publicly, as the crash dump contains a raw snapshot of the printer's memory and may include unencrypted sensitive information.

**Video**
  Nothing to see, it prints as usual.

bkerler, 2024-03-01:
Is there a specific reason why you use M486 S0 for canceling objects right after M600 ? I don't see any wipe tower or brim being used. Actually, I think this looks like a PrusaSlicer related issue. I do not see a color change in your 3mf project file nor in the bgcode file happening and also in the 3mf there are no virtual extruders are enabled, so I assume that something went wrong on slicing.

dwbmb, 2024-03-01:
What you say is very interesting, because if I open this 3mf in PrusaSlicer and bgcode in PrusSlacier G-code Viewer, there is pause print and color change present in both of them. And that is exactly why I posted this issue in firmware, not slicer issue page.

mix579, 2024-03-01:
As far as I know, the M486 commands were added to the gcode Prusaslicer creates when the Mini started to support cancel object. I'm pretty sure they have nothing to do with the OP's issue. I can't test the gcode as I'm away from home for a while but I'm quite certain I tried prints with color swaps on my Mini recently, so I don't know why this file isn't working 🤔

bkerler, 2024-03-01:
@dwbmb yes, that is weird. When I open up the PrusaSlicer 3mf, I see no color change. But as you said, opening the bgcode file, I can clearly see the color change happening at layer 506 and a pause at 123.

@mix579 yep, the M486 is unrelated to the issue, but still shouldn't be there imho.

Will dig deeper into the gcode then.

bkerler, 2024-03-01:
Just tested the gcode file, seems to work fine.
[image]

dwbmb, 2024-03-02:
Yes, this is what I was aiming for, but I wasn't that lucky. Tried twice, always got the single colour model without any pause. Do you have the same FW version as I do?

bkerler, 2024-03-03:
I tested with v5.1.2.

JahnSteven, 2024-03-22:
I have something similar happen to me today on 5.1.2.

I had a pause to change filaments but I ran out of filament instead (almost at the exact time). The change filament in the print wasn't yet reached. There was no way to tell as the printer was only ~1 layer away, but that is another issue.

I paused the print to cancel the object (since I put the wrong color on it), but when I went to change the filament back to the original color, the printer would not eject the wrong filament. Almost as if it wasn't unpausing itself to do so. The menu would just ask if the filament had been unloaded, and it was not unloading it.

closed 2024-07-02 by github-actions[bot]