prusa3d/Prusa-Firmware-Buddy #3785: [ENHANCEMENT] Position bed after finishing print in a way that it doesn't cover LCD

arekm, 2024-02-23:
**Printer type** - MK3.5
**Firmware** - 5.2.0

**Is your suggested improvement related to an existing problem? Please describe.**

When MK3.5 printer finishes print it moves bed to the front for easier print sheet removing from the printer.

Unfortunately with bigger LCD (in comparison to MK3S+) is goes too far into front in a such way that it starts to make top of the LCD not visible to the user.

No idea if that's MK3.5 thing or if it also affect MK4.

[image]

**Describe the expected functionality**

Don't move bed fully to front so it won't cover LCD. Like on this photo (bed was manually moved).

[image]

magistersmax, 2024-03-03:
This isn't a firmware issue, it's a gcode issue.  

You can fix this yourself by editing the end gcode in PrusaSlicer, a configuration update would be the way to roll it out to the community..

magistersmax, 2024-03-03:
For some more detail and how to fix this yourself, in PrusaSlicer go to the Custom G-code page, and look for this under End G-code:

G1 X241 Y201 F3600 ; park

That is the line that sets the position of the extruder/bed after the print is complete.  The Xxxx and Yyyy sets the X and Y position in mm.  I like my extruder to be on the left side of the axis and I agree that the bed should be back a bit more, so I just changed that line to this:

G1 X10 Y180 F3600 ; park

Save it as a user preset and it'll carry over to the next things you slice.

arekm, 2024-03-03:
Ok, Prusa devs, please move this end_gcode issue to https://github.com/prusa3d/PrusaSlicer-settings repository

Tupson444, 2024-03-04:
> For some more detail and how to fix this yourself, in PrusaSlicer go to the Custom G-code page...

Actually, that would not work in this case, because the printer will override the gcode (that is, it will execute the gcode, then it will move the bed on the default position, so the end gcode will not have an effect). This was reported here: #3790

> This isn't a firmware issue, it's a gcode issue.

While this indeed is a gcode issue, it cannot be solved without first solving the firmware issue (#3790).

magistersmax, 2024-03-04:
I stand corrected.  It does seem like #3790 is the actual issue here.

Acidham, 2024-06-11:
Any update before it gets closed?

danopernis, 2024-06-11:
This is certainly possible to fix with proper configuration of the Marlin NOZZLE_PARK_FEATURE

Acidham, 2024-06-11:
Agreed, and even better solution to add it to config. When the LCD is assembled to the printer, the heat bed is hiding the screen.

Tupson444, 2024-06-11:
> NOZZLE_PARK_FEATURE

This is it then, probably. The printer parks nozzle (print head) every time after finishing print.

This should be changed that it only parks if such a command is actually present in the gcode file (`G27` ?). Hardcoded commands hinder customizability.