prusa3d/Prusa-Firmware-Buddy #3789: [ENHANCEMENT] Ability to add SteelSheet selection in Footer line

Acidham, 2024-02-24:
Enter what type of printer or upgrade the enhancement applies to.
Printer type -MK3.5
MMU Upgrade - no

Is your enhancement related to a problem? Please describe.
When printing on the new MK3.5 it is important to directly see which steel sheet is being configured (for z-settings) to avoid damage.

Describe the solution you'd like
The ability to add selected steel sheet in footer line.

SnowB0und14, 2024-02-26:
Yes please, I damaged my PEI sheet as it’s significantly thicker than the satin sheet.  There’s no way to verify it kept your sheet selection.

ChrisHeerschap, 2024-03-27:
Came looking for a request to rename the sheets and saw mention of this so adding my name to the request for this as well.

jnovak-netsystemcz, 2024-04-01:
I'm missing this feature too.

I noticed there is FOOTER_HAS_SHEETS symbol which enables ability to select profile sheet name as one items showable in footer. The symbol is enabled just for PRUSA_MINI (see src/common/footer_def.hpp). Therefore it looks that enabling this symbol for other models (e.g. MK3.5) will do what is needed. But I have no way how to test it - I'm not able to compile own firmware to test it.

Can someone else try it?

lezetuser, 2024-04-02:
@jnovak-netsystemcz 
Not sure, that only adding this parameter will be sufficient. The GUI menu to select the parameter to be displayed and procedure displaying the particular parameters within the footer have to be extended as well.

lezetuser, 2024-04-03:
Good news: I had successfully compiled the code. Bad news: I do not have MK3.5 If you want to load/test the unofficial/unsigned FW you have to break the appendix (losing the HW/FW warranty) which means you do everything on your own risk/cost.

LukeShootsThings, 2024-05-12:
I don't know how else to show support for a given feature request beyond commenting, but I would love to see this feature be implemented for my MK3.5.

CZDanol, 2024-08-26:
Implemented in the 6.1 firmware.

closed 2024-08-26 by CZDanol