prusa3d/Prusa-Firmware-Buddy #3790: [BUG] Changing end-code has no effect after print finished

Acidham, 2024-02-24:
**Printer type** - MK3.5
**Printer firmware version** -  5.2.0+13881

**SD card or USB/Octoprint**
  USB and PrusaConnect

**Describe the bug**
 Changing End-Code has no effect
  
**To Reproduce**
  I tried to change Y value of following line `G1 X241 Y204 F3600` to prevent heatbed to hide the new XLCD. I changed to Y181. After finishing the print the heatbed still stops at Y204
  
**Expected behavior**
 Heatbed arrives at Y181 to unhide the LCD
  
**G-code**
 G1 X241 Y181 F3600

**Video**
 
[image]d
  Please send the crash dump file to Prusa by emailing it to reports@prusa3d.com. Sharing this file is important and helps us investigate the bug.

  Do not share the file publicly, as the crash dump contains a raw snapshot of the printer's memory and may include unencrypted sensitive information.

**Video**
  Please attach a video. It usually helps to solve the problem.

Acidham, 2024-02-29:
> (I think it just lifts as instructed by gcode, then lifts again for its default value, but I'm not sure)

Yes, I experienced exactly the behavior you described after change end-code for Y-Axis

bkerler, 2024-02-29:
Not sure if this is really a bug, I do assume that this is gcode vs. default printer movement handling after the print.

Acidham, 2024-06-11:
Any update before it gets closed?

danopernis, 2024-06-11:
This is not a bug, this is Marlin behaviour when using NOZZLE_PARK_FEATURE

I am closing this, but will keep https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3785 open

closed 2024-06-11 by danopernis