prusa3d/Prusa-Firmware-Buddy #3793: [BUG] Reproducible crash 17202 (Extruder Preheat Error)

smurfix, 2024-02-26:
**Printer type** - XL
**Printer firmware version** - 5.1.2
**Original or Custom firmware** - Original
**Optional upgrades** - no
**Describe the bug**
  I got #17202 despite the heat sensor working perfectly.

**How to reproduce**
  Select any extruder
  Set pre-heat to PETG
  Wait a few seconds (someties 20, sometimes less)
  Printer crashes and displays #17202 error

**Expected behavior**
  No crash.

  This cannot be a wiring error because it happens with every extruder. Also, the displayed temperature does increase normally.

**Crash dump file**
  not available, no dump file on memory stick

**Video**
  can do if required but probably not helpful

bkerler, 2024-03-10:
Did you contact support ? This clearly sounds like defective hardware.

smurfix, 2024-03-10:
Well, I have no idea which hardware that should be since printing itself works perfectly.

bkerler, 2024-03-11:
Let's put it that way: If something works for any other user but not for you, it's likely not fw related :)
Just because printing works fine doesn't mean that a sensor isn't damaged. Please contact support, thanks :)

smurfix, 2024-03-11:
I hesitate to speculate which sensor could possibly be *that* unused.

Anyway. I did find a hardware problem, this morning. The immediate culprit was a Heisenberg connection (i.e. it works as soon as you look at it) of the heater in extruder four. (Un)fortunately, I do have a spare hot end. ('Unfortunate' because the time hunting this problem should have been spent building a MK4.)

This issue is still a software problem, however, because it'd **_really_** help if the firmware told us (on the red screen) *which* sensor caused the crash! That information would have saved me two entire days of unscrewing and unplugging things. IMHO this is essential and warrants higher priority than than a "normal" feature request. The printer has too many sensors to leave this to guesswork.

Also, the fact that the crashes did happen when doing semi-unrelated things (like pre-heating extruder 1 or 2 … WTF?) does suggest that something else might also be amiss here.

bkerler, 2024-03-14:
@smurfix I will have a look if additional info can be added to the message :)

bkerler, 2024-03-16:
@smurfix I just had a look. If you have the serial port connected, it actually tells you which extruder is affected.
And also the lcd display does show the heater as a error code: E1 is extruder 1, E2 is extruder 2 etc...

closed 2024-07-02 by github-actions[bot]