prusa3d/Prusa-Firmware-Buddy #3796: [FEATURE REQUEST] Enable negative z-axis movement

cdmack, 2024-02-28:
If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - [MK4]
**Optional upgrades** - [Enclosure]

**Is your feature request related to an existing problem? Please describe.**
No

**Describe the expected functionality**
Upon powering up the printer, then preheating or loading the filament, the z-axis rises. This itself is not an issue per say. However, for those of us with enclosures, it can be troublesome. For example, if the axis raises and taps out at the top, it makes it difficult to load filament in some enclosures. The only way to move the z-axis down is to home the z-axis. Now for the feature request...Allow negative z-axis movement from the move axis menu in settings. We can only start at 0 and go up from there. With the loadcell sensor, I see no reason we cannot have negative values and the axis stops moving if it detects a load on the sensor. Homing the each axis is certainly a solution, but takes much more time.

cdmack, 2024-03-01:
@bryn51 I guess I never realized it on my MK3 since it is not in an enclosure so it wasn't a concern. I agree though. Whatever feature request and/or bug gets addressed first though would make me happy. Maybe post/link the bug to this so we can kill two birds with one stone. I am having trouble finding it.

bkerler, 2024-03-01:
I'd say it's a feature request. The MK4 printer has a X_MIN_POS of -1 and a X_MAX_POS of 251, same for Y_MIN_POS -4 and Y_MAX_POS of 211 and a Z_MIN_POS of 0 and a Z_MAX_POS that can been measured and stored into the eeprom. Now normally the homing process is used for calibrating the offsets and before homing this isn't possible as the printer doesn't know its position. That's why it is no bug, it's how it works.

Here's how it is being handled: https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/5fc26ee526a57b54ee5583100908522c2b3bb4b1/src/gui/menu_vars.cpp#L71-L83

So, to overcome this issue, it would require to ignore the ranges if homing didn't happen yet. As this is safety related, I will have a chat if a "override" is really what we want here or if any harm to the printer could be caused.

bryn51, 2024-03-01:
Here's something to consider. In this circumstance the extruder movement in any axis is controlled by the user  using the control knob.
The user has to be looking at the printer, thus unlikely to deliberately drive the extruder into the bed. Of course it's possible to turn the knob swifter than the printer can move, and it eventually catches up, might hit the bed by mistake. 

A safeguard could be added for those printers having Nextruder: the movement could be monitored using the loadcell. It could shut off movement if loadcell fires to act as safety control.

tkircher, 2024-03-07:
@bryn51 To add to that, when the user is moving the Z axis manually with the knob, it should immediately stop as soon as the knob stops moving, no matter how much movement is queued, like a regular CNC control pendant. Otherwise you don't really have any control over how far it moves or whether it smashes into something.

closed 2024-07-02 by github-actions[bot]