prusa3d/Prusa-Firmware-Buddy #3797: [BUG] Tool offsets calibration fails on on 6.0.0-alpha (works on 5.1)

m00dawg, 2024-02-28:
**Printer type** - 2 Head XL

**Printer firmware version** - 6.0.0-alpha

**Original or Custom firmware** - Original

**Optional upgrades** - 2 Head

**USB drive or USB/Octoprint**
  USB flash drive

**Describe the bug**
I flashed 6.0.0-alpha and immediately ran the new phase stepping calibration so I'm not sure if that was a factor or not. I then ran a tool offsets calibration as I also needed to retighten the rear screws that hold the toolheads. The initial touch on the top of the target worked, but then the toolhead moved too far to the right and never came back in contact with the target. Instead the printer paused, rebooted, and flashed a red error screen. Reverting to 5.1 worked (I was able to run the tool offsets cal without issue).

**How to reproduce**
1. Upgrade to 6.0.0
2. Run phase step cal
3. Then try to run tool offset cal

**Expected behavior**
The offset cal should work without issue as it does in 5.1

**G-code**
N/A

**Crash dump file**
N/A

**Video**
Ooops sorry, I already reverted back. I'll be aware of needing a video in the future.

Enematik, 2024-02-29:
Same issue on my XL 2T

BillC58, 2024-03-01:
I found that the plastic wheel on the calibration pin caused this issue with me. Once I removed it from the pin the calibration worked fine.

jacobfrasca, 2024-03-01:
To reproduce on my 2 Toolhead, I just upgraded and redid all the calibrations. The phase step calibration was not required to be completed to reproduce.

rdgillis, 2024-03-01:
Same issue as described in the original post.

I almost seems as if the move after the first pin tap is happening in the wrong direction, therefore not missing the “side-tests” of the pin

Video link is below.

https://share.icloud.com/photos/0caZ7jmsPHJSRemNpVcwK9Xjw

Aurelio67, 2024-03-02:
I have the same problem (XL 5T) but the initial touch on the top of the target worked, head moved to the right, then returned correctly but went over the target and not touch.
It seems that the plate after the first touch over the sensor went down.
I skipped this test and for the time being everything seems to be working properly, I noticed that it effectually makes less noise than before

Thanks

mtaumike, 2024-03-02:
ha, wish I would have seen this yesterday.

I needed to do some maintenance on TH4 cause .. things happened.

After doing my work on the nozzle/block I got everything reinstalled and ran calibration. I did this 5 times with at least 2 different errors showing up. Unstable probe and something about the nozzle not being round (from memory).

Reverted and ran things and clean as a whistle.

Essbante, 2024-03-07:
Reporting the same issue 2T to 5T upgrade, running 6.0.0 firmware. Crashes with error PIN NOT REACHED, [17107](https://help.prusa3d.com/article/pin-not-reached-17107-xl_470951)

Juan-VV, 2024-03-09:
I am seeing same issue - touches pin once in the middle, lifts a little and moves away, then goes back to search for pin horizontally and errors out.

Now looking up how to roll back to 5.1.2

kvtorp, 2024-03-09:
I just ran the tool offset calibration, and did not see this problem. However, the tool offsets were weird, the X offset was generally good, the Y offset on TH2 was fine, on TH3 a bit off, on TH4 very off, and on TH5 more than a millimeter off on Y. For all, I had to reduce the offset (make it more negative), all errors were in the same direction

WarrenSchultz, 2024-03-10:
Ran into this issue as well just now. 5T, offset calibration fails to reach the pin error.

wafflecart, 2024-03-10:
In to report the same issue, was driving me crazy! At the very beginning of calibrating a tool offset it first touches the top of the pin, next bed moves down and the head moves to the right, bed moves **DOWN AGAIN** instead of **UP**, thus misses the side pin probing. Then get error "PIN NOT REACHED".

I couldn't seem to get out of this loop **unless I did the entire calibration tests from scratch again.** 
 
This is how I replicate:
- start calibration tests from scratch but when it gets to the tool offset calibration make it fail the first time e.g with dirty nozzle or bit of filament to cause the "nozzle not being round" or if printer happens to crash during the calibration (often does..)
- go into calibration to redo just the tool offset and it does the above behaviour and gets "PIN NOT REACHED".
- try again and will still do it
- try redoing x,y,z or probe calibrations before doing tool offset calibration again and same result.

norwootj, 2024-03-13:
Same issue on my XL 2T

braw, 2024-03-26:
> I just ran the tool offset calibration, and did not see this problem. However, the tool offsets were weird, the X offset was generally good, the Y offset on TH2 was fine, on TH3 a bit off, on TH4 [...]

I just experienced this as well. I had 0.4 nozzles on using the calibration from 5.1.2, switched to 0.6 nozzles to print something, the calibration crashed, then successes. I swapped back to 0.4, ran the offset cal, and tried printing a PrusaBox that I was able to print perfectly before the swap and re-cal, and the hinges were off and fused. I ran the manual calibration and found all my Y axis were off, but X were fine. Manually calibrated the offset and trying again.

So aside from the crash, the offset cal does not appear to be accurate in the alpha.

aschmitz, 2024-03-29:
I ran into issues with tool offset calibration today on 6.0.0-alpha as well, although I had a somewhat different error than most people are experiencing: I received a "Tool offset out of bounds" error as pictured here:

[image]

(Transcription: "Tool 268462492 offset ?: 0.000 is out of bounds [5.031e+180, 0.000]." / "More detail at https://prusa.io/17104".)

My assumption is that some amount of memory is either not being initialized properly or is being written out of bounds, as that "tool number" is 0x1000699C, which feels like it's at least plausibly a memory address, rather than "tool 1-5". (Those bounds are also questionable, of course, but I haven't tried to figure out what a similar address would look like as a floating point number.)

I don't have a perfect memory of what happened to trigger this, but I believe it was:

1. The day before, I attempted to calibrate the tool offsets, but two tools failed because the nozzles weren't clean.
2. Today, I turned the printer on.
3. Heated extruder 3 to approximately 230° for manual cleaning.
4. Used the preheat menu to cool down.
5. Used the "Control" menu to trigger tool offset calibration, declined the option to heat the toolheads, allowed it to probe the sheet, removed the sheet and inserted the calibration pin, and continued the calibration.

I stepped away from the printer for quite some time and came back to it beeping and having all of the toolheads parked (and I do not remember the position of the toolchanger at that time), so I don't know at what point in the probing it failed, but I know that it had started with the first toolhead's probing by the time I left.

Memory corruption somewhere in the calibration process might explain the variety of different issues people have been seeing, but I'm not aware of a good way to trigger a dump of memory on demand, and didn't do so. Downgrading to 5.1.2 and running the calibration there worked fine.

dpark12300, 2024-03-30:
I had the same issue. #3873

dpark12300, 2024-03-30:
Also BSOD'd when i rebooted. I will send crash dump to reports@prusa3d.com soon.

wafflecart, 2024-03-31:
> I had the same issue. #3873

Search before and add your findings to any existing tickets before opening new ticket.

IntegersOfK, 2024-04-01:
> In to report the same issue, was driving me crazy! At the very beginning of calibrating a tool offset it first touches the top of the pin, next bed moves down and the head moves to the right, bed [...]

That's the exact behaviour I'm seeing. I decided to do this calibration with the new firmware because my first layer is about a millimeter too high on all my prints all of the sudden.

tg73, 2024-04-04:
I had the same behaviour. Removing the black plastic nut allowed the calibration to almost complete (5 tool), but the 5th tool complained about being non-circular. Visually, it looks like the side contact point is much higher up the nozzle tip than in 5.1.2. I will revert to 5.1.2, perform the calibration, then go back to 6.0.0-alpha - the quietness is just too nice to give up :smile:

eltron247, 2024-04-06:
Having the same issue. 

Fresh update to 6.0.0 alpha from 5.1.2, then going through the calibration from step 1 the XL will fail on tool offset calibration.
Printer grabs the first tool. Asks for a heatbed sheet to be installed. (It is) click continue.
Printer probes the nozzle against the sheet seemingly successfully then asks for the sheet to be removed and the cal pin installed.
Sheet is removed and pin is installed. Click continue. Printer probes the pin center, seemingly successfully. 

It then drops the bed and bring the carriage right from center. It does not raise the bed. The it trucks the carriage left until about 75% over the pin bypassing center because the bed is still lowered. THe printer seems to bsod crash and once rebooted issues the pin missed error with a red screen and beeping.

It does this loop continually.

danopernis, 2024-04-17:
Hi folks, can you please test using the latest release? Tool offset calibration should be fixed.

https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.0.0-RC3

Aurelio67, 2024-04-17:
Hi, i installed the latest version and the calibration was successful on all 5 heads.
I will try to do a test print later to see if everything is okay.
Good job.

Many Thanks

kvtorp, 2024-04-17:
Also ran tool offset calibration, as I switched several nozzles.
The alignment is **perfect**. At previous alignments, even with stable firmware, there was always a residual error that had to be removed manually, now no deviation from perfect can be measured on any nozzle. Did you do more than just fix it?
(I used [this](https://www.printables.com/model/761477-nozzle-alignment-aid) to check)

braw, 2024-04-17:
I ran the calibration again, and used the gcode from [here](https://help.prusa3d.com/article/multi-tool-manual-calibration-xl_470560).

Visually everything looks perfect from the gcode test print, but the numbers did change slightly. Maybe more accurate?

T2 (5.1.2):
X = 0.031799
Y = 0.019012
Z = -0.038177

T2 (6.0.0-RC3):
X = 0.060791
Y = 0.054474
Z = -0.043537

---------------

T3 (5.1.2):
X = -0.213654
Y = 0.150360
Z = -0.263859

T3 (6.0.0-RC3):
X = -0.110672
Y = 0.114929
Z = -0.267982

---------------

T4 (5.1.2):
X = -0.126877
Y = 0.215454
Z = -0.148987

T4 (6.0.0-RC3):
X = -0.075150
Y = 0.197647
Z = -0.146703

---------------

T5 (5.1.2):
X = -0.179062
Y = -0.534790
Z = 0.140838

T5 (6.0.0-RC3):
X = -0.133072
Y = -0.557770
Z = 0.136324

Juan-VV, 2024-04-17:
Install and tested, passed calibration, and did a long print off it. Did not see problems I saw previously with the alpha.

LOVING the reduced noise!!

closed 2024-04-18 by danopernis