prusa3d/Prusa-Firmware-Buddy #3799: [BUG] XL x/y carriage crashing when already in home position

muesli, 2024-03-01:
**Printer type** - XL 5-head

**Printer firmware version** - 6.0.0-alpha

**Original or Custom firmware** - Original

**Optional upgrades** - None, stock configuration

**USB drive or USB/Octoprint**
  USB flash drive & Prusa Link

**Describe the bug**
  When the x/y carriage is already homed, starting a new print will cause the carriage to crash into the frame.

**How to reproduce**
  I was about to start a print (sent via Prusa Link), and just while the carriage was homing, I realized I made a slicing error. I clicked "Stop" on the printer's screen, which caused the carriage to finish its homing process first and remain in home position. The printer then aborted the print correctly. I re-sliced my model, sent it to the printer via Prusa Link again, and went back to the printer to see it starting the new print. At this point the carriage tried to start its homing process while still being positioned in its home. This caused the carriage to rattle into the frame for a short while, before successfully recovering.

**Expected behavior**
  I expect the printer to realize the carriage is still homed and either skip homing or start the homing process without crashing into the frame first.

akamakamak, 2024-03-05:
I had the same issue with the latest release of the firmware.

tg73, 2024-03-12:
Probably related to #3724. Generally, it seems that stopping a print during the firmware-controlled parts of the pre-print sequence (homing, MBL, nozzle cleaning, possibly absorbing heat) is likely to leave the printer in a poorly-defined state. Until these issues are fixed, I would suggest hitting the reset button after the print has been stopped in this way, then at least the printer will (hopefully) not make any false assumptions about its state.

tg73, 2024-06-11:
Waiting for Prusa to respond.

tg73, 2024-09-02:
I suggest that Prusa confirm if this is covered by #3724, and if so, close as a duplicate.

closed 2024-11-12 by github-actions[bot]

muesli, 2024-12-05:
Still waiting for an official response.