prusa3d/Prusa-Firmware-Buddy #3801: [BUG] No option for accessible settings change

dawidpieper, 2024-03-01:
**Printer type** - all

**Printer firmware version** - all Buddy

**Original or Custom firmware** - Original

**Optional upgrades** - N/A

**USB drive or USB/Octoprint** Octoprint

**Describe the bug**
With end of sale of Prusa I3 MK3S+, I think it's a high time to present serious accessibility issue regarding new firmware.
In legacy Prusa-Firmware we had D-codes and ```M500``` enabling blind users to change any printer setting. I just could find a corresponding EEPROM address in the code and that was all.
With new Prusa-Firmware-Buddy I'm, as a blind person, unable to change basic settings. For example we spent some weeks ago over a hour on your Live Chat finding any way of storing Z-offset to EEPROM - no success.
If Prusa MK4 is to be successor of Prusa I3 MK3S+, please don't make such accessibility regression.
We don't specifically need return of D-codes, but any way to change printer settings via USB.

**How to reproduce**
N/A

**Expected behavior**
Any possibility of changing settings without access to the screen.

**G-code**
N/A

**Crash dump file**
N/A

**Video**
N/A

dawidpieper, 2024-03-02:
I just thought to myself that we probably have a solution, in addition relatively simple to implement and also useful for other users during debug or configuring print farms.
Since we can use a txt file to configure the network, we could expand this idea to export/import all settings from the printer.
A simple gCode could be created that exports and imports settings from a txt file to a flash drive.

bkerler, 2024-03-10:
As this is no bug, please change the title to feature request. Thanks ;)

dawidpieper, 2024-03-10:
Hi @bkerler
If you had an option to use your device and it disappeared in the next release, you'd consider it a bug.
We had accessibility options in legacy firmware, and since MK3.5/MK4 are officially said as successors to MK3 series, I consider lack of possibility we had in previous model is a bug. :)

tkircher, 2024-03-25:
As @dawidpieper said, this is properly a regression, not a feature request.