prusa3d/Prusa-Firmware-Buddy #3804: [ENHANCEMENT] Adding STEALTH variations of print profiles

jonnieZG, 2024-03-03:
**Printer type** - Prusa MK4

**Is your suggested improvement related to an existing problem? Please describe.**
In some cases printing is too loud for night prints, especially non-printing (Travel) moves of the print bed.

**Describe the expected functionality**
Limiting travel speed to 80 mm/s and travel acceleration to 2500 mm/s² makes printer significantly quieter.

The simplest way (that I currently use) is that I added STEALTH variations of STRUCTURAL and SPEED print profiles, but adding a simple chechbox in PrusaSlicer for quick enabling/disabling such a STEALTH mode (that would automatically adjust said speed and acceleration), or perhaps even returning the STEALTH mode back into the printer menu would be very useful.

closed 2024-04-19 by jonnieZG

TimetravelerDD, 2024-06-24:
Thanks! could we also get a toggle in Prusa Connect to switch this on/off ?

danopernis, 2024-06-24:
@TimetravelerDD it would be better to open separate feature request instead of commenting the closed one.