prusa3d/Prusa-Firmware-Buddy #3807: [BUG] Print head still moves to cancelled objects

andywarburton, 2024-03-04:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK4

**Printer firmware version** - 5.1.3+13503

**Original or Custom firmware** - Original

**Optional upgrades** - Stock (built from kit)

**USB drive or USB/Octoprint** USB / Prusa Connect

**Describe the bug**

When printing with multiple objects, I have noticed that if you cancel an object the print head will still move to the object before not printing it. This is a minor thing but I have noticed it causes some stringing and has resulted in pulling over non-cancelled objects due to the stringing.

**How to reproduce**
I printed the allen key holder from printables in PETG. For some reason one of the small side pieces came loose from the bed so I cancelled it but noticed that the head still kept moving to the cancelled object, creating stringing.

**Expected behavior**
Don't move the head to the 

**G-code**
(Github won't allow me to attach gcode to an issue! so I renamed it as .txt)
[vase_0.4n_0.2mm_PLA_MK4_3h19m.gcode.txt](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/14484243/vase_0.4n_0.2mm_PLA_MK4_3h19m.gcode.txt)

**Crash dump file**
n/a

Fluffybunny96, 2024-03-07:
I'm having the same issue, though it was odd but just let it go. Running FW 5.1.3

murk-sy, 2024-03-08:
This is (I believe) intended behaviour, as skipping travel moves can cause the nozzle to crash into non-skipped parts

[image]

I believe this is mostly resolvable (with some caveats for special cases), but it would have to be implemented in the slicer, rather than the firmware.

closed 2024-07-03 by github-actions[bot]