prusa3d/Prusa-Firmware-Buddy #3808: [ENHANCEMENT] FW 6.0.0-alpha

chmuc01, 2024-03-04:
Prusa XL - FW 6.0.0-alpha 

The phase stepping calibration of the motors workes very good and the printer is really quiet now as long as it is moving fast.
But **for the external perimeters the print speed** (Prusa Slicer settings: XL, 2 heads, IS, 0.2mm structural) i**s slow and the motors still resonate a lot**. This ends in still having VFAs where they should not be: at the visible outside of a part.

Maybe it would be a good idea to also find a phase stepping calibration for these slow speeds, too, as this may really reduce the VFAs.

yaqwsx, 2024-03-10:
Phase stepping should eliminate any resonance in slow and mid-range speeds. If this is not the case for you, could you:
- share the numbers reported by the phase stepping calibration and
- ensure that you have phase stepping enabled in settings and it is not disabled by accident?

chmuc01, 2024-03-16:
Hello Jan.
Die reduction that was displayed was between 94 and 97%.
And of course I switched phase steppin on.
I realize that the sound level is really gone dramatically down now,
otherwise I wouldn't have noticed.
Normally I do not hear any movements any more because now the noise is
below the noise of the fans :-).
But as soon as the external perimeter is printed, I could hear a very
noticable change in the sound level with vibrations that are nearly the
same as before.

As I have just updated from the 2 head to the 5 head version, I will do
another test now.

yaqwsx, 2024-03-16:
Could you provide a video with a sound recording? From the vague description, it is nearly impossible to tell what is wrong.

chmuc01, 2024-03-25:
Seems as the normal settings for external perimeters are so slow that the
noise and vibration comes from the slow stepping of the motors.
So I think there is no problem in the phase setting rather just in the slow
motion.
Maybe it would be interesting to change some of the printing profiles in
the configuration for PrusaSlicer to just get a faster motion as it is
unnecessarily slow for external parameters. In the past I was using the
default settings from PrusaSlicer but now I have changed the speed for the
external parameters and now it is not as noisy.

closed 2024-07-30 by github-actions[bot]