prusa3d/Prusa-Firmware-Buddy #3809: [FEATURE REQUEST] <Implement Hostname for PrusaLink or at least DHCP with static IP address>

mjbstldc, 2024-03-04:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - [MK4]

**Is your feature request related to an existing problem? Please describe.**

User cannot address wifi connected printer by hostname, even if you put specify hostname in the configuration file. 
This means you have to configure Prusaslicer by IP address, and this changes due to DHCP.

An alternative is to allow DHCP with a reserved IP address. Most home routers will allow this, but the IP address in the file is ignored if type = DHCP

I am told you can set a static IP address for Wifi by editing the file, but many home routers will not permit this. If you want to set a reserved IP address it still uses DHCP. 

**Describe the expected functionality**
Add hostname to the PrusaLink path. 
or
enable reserved IP address to be used by DHCP.

beelsebob, 2024-03-05:
mDNS support would be awesome.

bkerler, 2024-03-10:
Not sure if you posted this issue in the right repository, this is Xbuddy FW related, for Prusa-Link there is a different repository.

mjbstldc, 2024-03-10:
The changes would be in the firmware.

bkerler, 2024-03-10:
@mjbstldc I still don't see why this has to happen in the firmware:

"This means you have to configure Prusaslicer by IP address, and this changes due to DHCP." => This is Prusaslicer related and handled in the Prusaslicer repository.

"I am told you can set a static IP address for Wifi by editing the file, but many home routers will not permit this. If you want to set a reserved IP address it still uses DHCP." => There is already an option to set either static IP or dhcp IP. DHCP isn't handled in the printer, but in the dhcp server instead, which is normally your home router.

"Add hostname to the PrusaLink path." => That is part of PrusaLink, not of the printer firmware.
"enable reserved IP address to be used by DHCP." => Done in your router, not printer firmware.

bkerler, 2024-03-10:
@beelsebob I might have a look into mDNS as part of adding ipv6 support

beelsebob, 2024-03-10:
Nice!  That would solve all my IP knowing issues.

bkerler, 2024-03-10:
@beelsebob Sorry to say, but I had a look and it seems that the code for mdns has been removed by Prusa (see: https://github.com/prusa3d/LwIP_212), so I cannot add mdns support for now :(

mjbstldc, 2024-03-10:
Could you allow DHCP with manual IP address? That’s not as elegant as MDNS, but it is a workaround. A lot of wifi routers will force users through DHCP, but at least some will allow DHCP with manual address. See attached img and thanks for considering!

￼

beelsebob, 2024-03-10:
> @beelsebob Sorry to say, but I had a look and it seems that the code for mdns has been removed by Prusa (see: https://github.com/prusa3d/LwIP_212), so I cannot add mdns support for now :(

I mean, doesn't that just mean that any patch to add support would need to put the code back in?

bkerler, 2024-03-11:
@mjbstldc I don't think it's that trivial to do that, as lwip sometimes can be very painful, but I look into it.
@beelsebob Adding the code back is of course possible, but I assume it was removed for a good reason....

mjbstldc, 2024-03-11:
Maybe. But Prusa sells a lot of printers to people who are non-technical based on how easy they are to use. When they see wifi advertised, they may expect it will be easy to connect.

vkorenev, 2024-03-13:
@mjbstldc, I think the description of this feature request needs to be changed. What Prusa developers can do and should do is to support mDNS. This will allow to discover the printer (and PrusaLink running on it) by a domain name.

If you want "static" DHCP address, that can be done on DHCP server (i.e. your home router) side only. Many routers can be configured to assign a static address to a certain MAC address. However, AFAIK a DHCP client cannot request that.

vkorenev, 2024-03-13:
> Not sure if you posted this issue in the right repository, this is Xbuddy FW related, for Prusa-Link there is a different repository.

@bkerler, I might be mistaken, but my understanding that Prusa-Link backend (REST API) code is in this repository. However, the issue is not in Prula-Link itself. It is utterly annoying that there is no stable way to address the printer (and thus Prusa-Link). For example, with the default settings, the DHCP server on my router assigns a new IP address to my printer every time I turn my it off and then on.

mjbstldc, 2024-03-13:
If mdns is feasible, it is the better solution. I was just trying to come up with a workaround that is not difficult, since they pulled the code. If you want to delete this feature request and write a proper mdns feature request, that’s fine by me.

bkerler, 2024-03-13:
As I am currently adding ipv6 support, I also fully ported the prusa lwip to the latest version 2.2.0 (see here: https://github.com/prusa3d/LwIP_212/pull/3). Now I have to wait for it being accepted and then I can discuss if I can also add mDNS. 

@vkorenev you should be also able to use the hostname of the MK4, which should be "PrusaMK4", instead of the ip. But your router should normally also be able to set a fixed ip address in the dhcp server settings, that's nothing a dhcp client could do. But you can set a static ip on the mk4 instead without using dhcp, that's already supported.

vkorenev, 2024-03-14:
@bkerler

> you should be also able to use the hostname of the MK4, which should be "PrusaMK4", instead of the ip.

I've just checked, this name does not resolve to an IP in my network. How is this supposed to work? Don't we need mDNS or something similar for this to work?

> But your router should normally also be able to set a fixed ip address in the dhcp server settings

Yep, that is what I ended up doing eventually. However, this was the first time in my life when I had to resort to this.

> that's nothing a dhcp client could do

Yes, I know: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3809#issuecomment-1993717275

> But you can set a static ip on the mk4 instead without using dhcp

Yes, but if using a router with DHCP, this static IP needs to be routable and out of the DHCP address range at the same time

bkerler, 2024-03-14:
mdns actually just uses multicast for propagation in the local network, which also requires router and end devices being capable of that (which isn't often the case), so you can use PrusaMK4.local for example as hostname without having a dhcp server. I'm no network specialist, but using my router, the router automatically detects PrusaMK4 as name and offers it in the network, so I don't need to use any ip address.

bkerler, 2024-03-14:
Just had a look, in src/common/config.h, LAN_HOSTNAME_DEF is set to "PrusaMK4" for the MK4. But you can change the hostname using prusa_printer_settings.ini as well. And that hostname is being used by dns servers.

bkerler, 2024-03-14:
@mjbstldc please change the topic to "[FEATURE REQUEST] mdns support", as this does suit more what is needed. Thx :)

amoose136, 2024-03-30:
I configure my router to assign static addresses but I can’t help but feel that I should be able to get to the printer with a “printerName.local” address as many users aren’t going to know anything about networking and doing something as simple as punching the printer’s ip address into PrusaSlicer and setting it to static on the router is going to be too much to ask so those users just are stuck with either using prusaConnect or sneakerNetting the usb drive every time they want to print something. I don’t  know what the right way to handle multiple of the same kind of printer being setup is (conflicts for the same hostname) but some way of using mDNS to get to the printer would be an excellent QoL improvement.

amiliv72, 2024-06-20:
I've asked about this feature here a while ago: https://forum.prusa3d.com/forum/english-forum-original-prusa-i3-mk4-hardware-firmware-and-software-help/setting-local-hostname-for-prusalink/

[mDNS](https://en.wikipedia.org/wiki/Multicast_DNS) is also known as Bonjour protocol (in OS X) and Avahi (in Linux). It is a zero configuration protocol described in [RFC 6762](https://datatracker.ietf.org/doc/html/rfc6762). In essence, it allows a device to advertise its name on the local network, so that it is accessible by name (e.g. you can access it via web browser as `http://some-name.local`).

Octopi supports this, and it is awesome feature. Since PrusaLink on the Mk3 with WiFi module was using Octopi under the hood, this worked in the past. However, the new PrusaLink implementation on Mk4 dropped this feature (for some reason).

The "ask your network admin to assign static IP and manually add DNS entry", or "edit hosts file on your computer" are very user unfriendly alternatives to mDNS.

amiliv72, 2024-06-20:
Hmmm... It'd seem this was implemented in 6.x.x firmware. But documentation for [setting up PrusaLink](https://help.prusa3d.com/guide/wi-fi-and-prusalink-setup-mk4-mk3-9-mk3-5-xl-mini_413293) was not yet updated. In particular it doesn't tell users to set the hostname in the network section of the file:

```
[network]
hostname = prusa-mk4
```

Once I added the above to the prusa_printer_settings.ini file and reloaded WiFi settings (under `Network > Wi-Fi > Load credentials`), it seems to work. I can access the printer using its local name (both in web browser and ping it from command line):

```
$ ping prusa-mk4.local
PING prusa-mk4.local (192.168.1.10): 56 data bytes
64 bytes from 192.168.1.10: icmp_seq=0 ttl=255 time=201.520 ms
64 bytes from 192.168.1.10: icmp_seq=1 ttl=255 time=6.917 ms
64 bytes from 192.168.1.10: icmp_seq=2 ttl=255 time=9.581 ms
64 bytes from 192.168.1.10: icmp_seq=3 ttl=255 time=14.421 ms
```

mjbstldc, 2024-06-20:
I gave up and went back to Octoprint. It’s a shame because other printers are becoming easier for ordinary people to use. That was a large part of the value proposition when the mk3 came out, but they need to continue to prioritize it IMO. The consumer writeups are now talking about Bambu Labs as the printer that just works, and Prusa as the printer for the hobbyist.

danopernis, 2024-06-20:
Hi, MDNS is supported since v6.0.0 via https://github.com/prusa3d/Prusa-Firmware-Buddy/commit/8a38e5a6845c16423840dcc86dccb8bd896f7263 I am not sure why it is not mentioned in the release notes, but it should be working now :shrug:

closed 2024-06-20 by danopernis