prusa3d/Prusa-Firmware-Buddy #3811: [BUG] XL 6.0-alpha FW - Z axis cannot move up

pugilist, 2024-03-05:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

Printer type: XL

**Printer firmware version** - 6.0-alpha

**Original or Custom firmware** - Original (unless alpha fw counts)

**Optional upgrades** - N/A

**USB drive or USB/Octoprint**
USB

**Describe the bug**
When the printer is powered on, the control panel seems to think whatever the z height was when the printer was powered on is at 0mm.  As such, you can lower the z-axis, but cannot raise it.  Additionally, when I discovered this, I was lowering the bed, and the printer made some awful noise at the bottom of the z-axis travel.  This almost makes me think that it was trying to lower the bed even more than what is actually possible.

**How to reproduce**
- Upgrade to alpha firmware (see above)
- Using control knob, lower z axis.  
- power the printer off then on again
- Observe that the printer reports z height of 0 and attempt to move the print bed up.

**Expected behavior**
I expect that the printer will properly understand the z-axis position and that the user will be able to move the bed with the printer's control panel

**G-code**
N/A

**Crash dump file**
I tried to save a crash dump, but none exists.  Let me know if this is useful and I can attempt to provide one :)

**Video**
will do.

pugilist, 2024-03-05:
Looks like this issue can be resolved by running the z calibration.

I also downgraded to FW 5.1.2 and found the same issue to be present.

murk-sy, 2024-03-08:
This is to prevent the bed crashing into a nozzle if a tool is loaded and axis calibrations haven't been run yet. It is essentially intended behaviour.

Though I do wish reaching max z (or min/max on most uncalibrated axes on other models too) would work without the scary stepper skipping noise. Something to do with crash detection and such as well.

If you regularly turn off your printer and wish to have Z at max when you turn it on again, you could change your print end gcode to always move the bed down after a print. Personally I don't see the appeal of that but it's an option.

pugilist, 2024-03-08:
Hey Murk-sy!

I suppose that makes sense.  In playing with it a bit further, I see that the printer will not move in the negative direction on any axis prior to running an auto-home.

I suppose that pushes this more in the realm of a UX issue, then.  I'd propose that either:

1) The printer warns the user before allowing manual control of the bed/gantry
2) The printer offers to run the auto home when the user attempts to manually control the bed/grantry

closed 2024-07-03 by github-actions[bot]