prusa3d/Prusa-Firmware-Buddy #3812: [FEATURE REQUEST] Add Ready Status as an option for ui Footer

Shinomorisama, 2024-03-06:
MK4 and probably mini and XL

It is currently possible to go to the control menu and set the printer as ready so it will grab the next item in queue or immediately start anything added to it's queue.  This is awesome and tremendous asset to productivity.   The issue is that the only way to tell if a printer is ready at the machine is to go into the control menu and see if the option is greyed out. Which is quite inconveniant.

The Feature I would request is something on the main screen to indicate the printer is ready.  This could be an icon on the top bar or as the title suggests and optional footer indicator.

murk-sy, 2024-03-08:
+1
See also / not really a duplicate but related #3315

ChrisHeerschap, 2024-04-29:
Alternately, having _any_ visual feedback when choosing "Set Ready" would be invaluable.

Shinomorisama, 2024-07-03:
Having worked with this more.  An indicator on the LED would also be excellent. Example: Red/yellow-errors or intervention needed depending on severity. Green-Ready or printing.  Blue-Not ready.

ChrisHeerschap, 2024-07-03:
> Having worked with this more. An indicator on the LED would also be excellent.

Yes, that would be excellent!

bkerler, 2024-12-14:
I just added a PR #4348 in order to add a footer with ready status

bkerler, 2024-12-14:
@Shinomorisama I recommend that your led feature request should be make as a new issue with "[FEATURE REQUEST] label :)

CZDanol, 2024-12-14:
Hello, I'm not sure if this is already out in 6.2.0-alpha2 or if will get out in the next firmware, but we are now displaying the connect status (plus ready status) in the header.

closed 2024-12-14 by CZDanol