prusa3d/Prusa-Firmware-Buddy #3813: [FEATURE REQUEST] XL Tool Light Dimming

rvalle, 2024-03-06:
**Printer type** - [XL]
**Optional upgrades** - [Multiple Tools]

**Is your feature request related to an existing problem?

There is a tool light that helps watching the print with naked eye. However, it blinds Cameras during remote watching. 
It is currently possible to turn on/off the LED: However turning the light of is sub-optimal too, as the shade of the tool head makes it difficult to watch the print too.

**Describe the expected functionality**

Ideally, Instead of tool light on-off, there could be a slider from ON to OFF in a reasonable number of steps (i.e. 10). So that the light can work best with the camera used for remote monitoring of the print.

mtaumike, 2024-03-29:
Even better would be the ability to control via GCODE

dpark12300, 2024-03-29:
Also, please implement this into Prusa connect with RGB control too.

KiitoX, 2024-06-13:
Still relevant I think

nealzip, 2024-08-21:
I think this is an important Option +1

legomind, 2024-09-09:
It would also be handy to have an api option for controlling the lights remotely

DoubleStrike, 2024-10-07:
As someone who recently started using OctoPrint, this is definitely still useful. Having the light on is just too bright for the camera, but off can be challenging to see progress on a complex print. Assuming this LED has PWM control, having a UI setting for it as well as a gcode command would be very handy for folks doing camera-based monitoring.

Cyberdal, 2025-04-15:
Does it work now to turn off or dim the led?

CZDanol, 2025-04-15:
Internal ticket: BFW-7078. We will propose this to the product guys, I think they will agree.

ianheat83, 2025-06-27:
Would also like this very much. Can’t use a camera near the nozzle effectively otherwise.