prusa3d/Prusa-Firmware-Buddy #3816: [BUG] <No heat after crash>

anandsathy, 2024-03-08:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [Prusa i3 MK3S+ with MMU3]

**Printer firmware version** - [3.13.3]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [MMU3]

**USB drive or USB/Octoprint**
  SD CARD

**Describe the bug**
  When an X, Y, or Z crash occurs, the printer does not heat up when I try to unload the filament into the MMU3.

**How to reproduce**
  Do an X, Y, or Z crash and select not to continue the print

**Expected behavior**
  The MMU will attempt to unload the filament even though the extruder is not heated

bkerler, 2024-03-10:
Please close this issue and reopen it in the right repository (Prusa-Firmware). This is for xbuddy boards (mk4/xl/mini)

anandsathy, 2024-03-10:
Hi,
Sorry that I posted the issue in the wrong place. I couldn't find the
regular firmware repository. I will change it now.
--
Anand Sathy

closed 2024-03-10 by anandsathy