prusa3d/Prusa-Firmware-Buddy #3820: [ENHANCEMENT] Jump to 200°C in Temperature Menu

TolleyLikesRice, 2024-03-09:
**Printer type** - XL, possibly MK4 & Mini

**Is your suggested improvement related to an existing problem? Please describe.**
Similar to what was added in MK3 Firmware 3.13.2, the temperature settings under the "Control" menu should start at 200/50, so it's quicker to change them. Very minor QoL improvement, but it's nice to have.

**Describe the expected functionality**
(copy pasted from 3.13.2's release notes, and changed a bit)

In the control menu, both nozzle and heatbed temperatures start at 0 °C, and users have to rotate the knob clockwise to increase the temperature incrementally by 1 °C until the target temperature was reached.

However, since most filaments require printing temperatures above 200/50 °C (nozzle/heatbed), this results in a lot of unnecessary turns. It should be that, starting at 0 °C, a single clockwise rotation will increase the nozzle temperature to 200°C instead of just 1 °C. The same change should apply to setting the heatbed temperature (0 °C to 50 °C). This modification allows users to reach their target temperature more quickly.

Similar behavior is should be applied to the print fan; when turning the knob clockwise from 0, the speed is set to 127, which corresponds to 50% of the RPM range.

If you prefer to revert to the original behavior, you should be able to simply rotate the knob counterclockwise first, and then you can use it in the same way as before.

brettvitaz, 2024-03-20:
I like this idea, though I would suggest activating it by rotating the dial counterclockwise when the temperature is 0 and leaving the normal operation in place.

wafflecart, 2024-04-08:
+1 for this, I feel like the rotary encoder on the knob is gonna wear out at this rate.. I went back to my mk3s recently and noticed how much better and quicker it was to set temps.

closed 2024-07-03 by github-actions[bot]