prusa3d/Prusa-Firmware-Buddy #3824: 6.0.0-alpha: Random layer shifts

gbzh01, 2024-03-12:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

XL 5 head

6.0.0-alpha

I experienced some layer shifts on that firmware. Since downgrading, they seem to be gone.

**How to reproduce**
The files had several copies of the same design.

**G-code**
 I can't say that it is G-code specific.

DoubleStrike, 2024-03-12:
I had one of these.  As you can see in this picture, the top layer shifted around 10mm to the lower right at a perfect 45 degree angle (which seems to indicate that only 1 of the 2 CoreXY motors was involved somehow).

[image]

danopernis, 2024-06-11:
Hi folks, the patch version 6.0.2 should contain some fixes to rare race conditions in phase stepping, can you test that?

closed 2024-08-20 by github-actions[bot]