prusa3d/Prusa-Firmware-Buddy #3826: [FEATURE REQUEST] Send Pushover Messages

MiOrt, 2024-03-12:
**Printer type** - [MK4] but basically all printers with network functionality 

**Is your feature request related to an existing problem? Please describe.**
  Continuous monitoring of MK4 PrusaLink necessery to see status of printer/print job.

**Describe the expected functionality**
  Implement Pushover integration with configurable messages. So e.g. "send message on job finished", "send message on change filament" (on multicolour prints), "send alert on crash detected" etc.

arekm, 2024-03-12:
Prusa has some notification integrations... but in Prusa Connect. And IMO doing such integrations there is a better place than bloating firmware.

Unfortunately no Pushover and even no generic https:// POST support.

ps. Prusa Connect website should really have a github repository (without code) only for issues. Same for printables.

MiOrt, 2024-03-13:
Well, I don't call that bloating firmware (same could be said for color-display or LED bar), I would say that is "keep things local as much as possible". I don't want to use Prusa Connect and really appreciate the Prusa Link integration. However if there is enough memory left, why not maximise user experience without cloud stuff.

amoose136, 2024-03-30:
These are new boards and memory usage for reasonable features _shouldn’t_ be a huge concern.

miltieIV2, 2024-05-13:
Assuming you don't want to go the PrusaConnect route (which works great BTW), how about the Home Assistant PrusaLink integration https://www.home-assistant.io/integrations/prusalink/ ?

samturner3, 2024-06-04:
+ 1 for pushover / pushbullet / generic https:// POST support via prusa connect. Is there a separate issues / repo for prusa connect specific discussion?

mgrn0, 2024-07-04:
+1 for pushover integration, too

mgrn0, 2025-01-13:
shall we open a new one or could this one be re-opened?

CZDanol, 2025-01-13:
Reopening

zehkaiser, 2025-01-15:
> Assuming you don't want to go the PrusaConnect route (which works great BTW), how about the Home Assistant PrusaLink integration https://www.home-assistant.io/integrations/prusalink/ ?

This is the route I went. You can monitor the printer status locally and send notifications to phones, tablets, or whatever you want.

mgrn0, 2025-01-15:
> https://www.home-assistant.io/integrations/prusalink/

does this also work with the mk4s? the last changes were committed 4 months ago,
was hoping for a more generic/independent approach with pushover.

zehkaiser, 2025-01-15:
It should work fine. While it doesn't specifically list the MK4S, the firmware for the MK4 is very similar, and includes the API that the integration relies on.

mgrn0, 2025-03-17:
I recently had a lot of times when my mk4s did send a notification through the native app but it arrived really late, between immediately and ~25 mins late. A pushover integration would help in these cases, too.

Prusa-Support, 2025-04-29:
Prusa Link should remain a most basic printer interface on the local network, especially because it is hosted on the printer's motherboard itself. It is intentionally only equipped with very basic features and most importantly, it is a bridge between the printer and Prusa Connect.

Because Prusa Connect is not an open project, its team is not really active on GitHub, even though we are receiving a few issues at https://github.com/prusa3d/Prusa-Connect-SDK-Printer.
The best would probably be sharing feedback via the special button always available on the bottom-right corner on [connect.prusa3d.com](https://connect.prusa3d.com/) website.
The Prusa Connect team collects feedback and requests through it, in a similar fashion to GitHub issues.
Here is the direct link. https://prusaconnect.sleekplan.app/?type=feature

The Prusa Link code, pretty much like the printer's firmware, is open though, and it is integrated in the printer's firmware, so you may access it for possible customization via the specific printer firmware source code or find the Prusa Link "core" source code in the Prusa Link specific repository (mostly dedicated to RPi version for MK3 8-bit firmware).
https://github.com/prusa3d/Prusa-Firmware-Buddy/releases
https://github.com/prusa3d/Prusa-Link/releases

Michele Moramarco
Prusa Research

closed 2025-07-09 by github-actions[bot]