prusa3d/Prusa-Firmware-Buddy #3827: [BUG] Firmware tags are not signed, recently

RichiH, 2024-03-13:
@JohnnyDeer I don't know if the release toolchain changed or if you simply forgot, but the last two release tags were not signed properly, leading to [6.0.0-alpha](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.0.0-alpha) & [5.2.1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.2.1) not being verified in GH UI:

## Tags
[image]

## Releases

[image]
[image]

and compare to

[image]

danopernis, 2024-04-16:
Thanks for the heads up, but this is actually not a bug. The "verified" icon by itself is quite meaningless, albeit it looks nice and green :smile: It is entirely depending on the settings of the actual developer who pushed the corresponding tag to this repository.

The firmware binary itself is always signed when it comes to official releases. The signature is verified by the bootloader during the boot.

closed 2024-04-16 by danopernis

RichiH, 2024-04-16:
> icon by itself is quite meaningless, albeit it looks nice and green 😄 It

I respectfully disagree with your assessment; it is a regression.

It's not about signing binaries, it's about signing the release tags. If
engineers forget/don't know how it's a process and/or training issue, in my
opinion.

Sent by mobile; please excuse my brevity.

danopernis, 2024-04-16:
Let's agree to disagree. Signing commits with GPG, creating tags referring to said commits and using github to create a release page was never promised as part of the contract. While I generally agree that it is a good practice, I certainly don't see it as neither a regression nor a bug.