prusa3d/Prusa-Firmware-Buddy #3830: [BUG] Running filament change after a print locks up the screen

murk-sy, 2024-03-14:
**Printer type** - XL 5 tool

**Printer firmware version** - [6.0.0-alpha](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.0.0-alpha)

**Original or Custom firmware** - Original

**USB drive or USB/Octoprint**
Prusa Connect/USB drive?

**Describe the bug**
After a print completes, running filament change appears to get the printer stuck at the hourglass. The tools appear to be heating up based on the fan sounds but even after waiting a few minutes, nothing happens.

**How to reproduce**
1) Run a multi tool print (from Prusa Connect?)
1a) Spool join was used twice when this error occured - this may or may not be related
2) After the print is done, in multi filament change menu, select a few tools for a filament change
3) The screen appears to freeze on the hourglass display

**Expected behavior**
Filament change proceeds normally

**G-code**
Cannot share the gcode, will try to create a new gcode to reproduce the issue

**Crash dump file**
Not sure if crash dump was created, will check when it's free

**Video**
Will try to reproduce, printer is occupied for a while

murk-sy, 2024-04-21:
Have since had another (or two? I really can't remember) instances when running multi filament change started, but the printer reset when the toolchanger reached (but not engaged) the first tool to change, and then reported a toolhead crash.

I'm still busy with that project, hoping to have some reproducible steps by the end of next week.

murk-sy, 2024-04-29:
Did not immediately reproduce on 6.0.0 stable, will reopen if it happens again.

murk-sy, 2024-06-11:
Reproduced on 6.0.1+14848, was likely the same on 6.0.0 as well

It's been a few days since the last print was finished on the XL so the issue can appear immediately after closing the "print finished" screen and after it's been closed for a while.

murk-sy, 2024-06-18:
May also be related to
- https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4040

Haven't tested since I've got a priority job right now

crasbe, 2025-01-17:
This is copied from a bug report I was just about to open, then I found this here. So it might be unspecific to this bug report in some areas.

<hr />

When I try to change the filament of the tool used after completing a long print (>12h), the printer is stuck in the "Please Wait" screen.
Usually the printer would start with the homing procedure and then perform the filament change.
However no movement happens and it's just the hourglass animation.

[image]

The printer beeps when pressing the button, but it does not do anything and the only way to exit is to press the Reset button. It does not appear to be frozen though, as the time is still running. It can be accessed via PrusaConnect, however the status is just "Idling": (I initiated the filament change at 15:59)

[image]

If I had to have a wild guess on what the cause might be, it looks like the State Machine of the Filament change does not even start. Perhaps the cause might be in this function: https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/136fb25a9f97434ea47b6c67b4d69eb66f83469a/src/gui/dialogs/DialogLoadUnload.cpp#L403-L470

It just looks like the state machine is not advancing from the first state to the state that starts the homing process. I was not able to find the part in the code where the homing is actually executed. There must be some place that accesses the current state, but I didn't find it yet.

<hr/>

Steps to reproduce the issue:

- Start the printer
- Change the filament in the tool you want to use (I am not sure if that is required, but that's what I know I did the last time)
- Start a print through PrusaConnect (the behavior does not change whether the file is already on the printer or was uploaded just before). I am not sure if the behavior is the same when starting the print from the USB drive, I'll have to try that.
- Let the long print finish (I don't know where the threshold is for the failure or if it's coincidental, however it could be reproduced on a 12-20h print, with and without tool change. I am not sure if that is every time or just most of the time)
- Change the filament in the tool used for the long print
- It should stay in the "Please Wait" screen indefinitely until Reset before doing any movements
- Reset the printer

I'm not sure if the issue occured before 6.2.0-alpha1 as well, I mostly started printing long projects after the upgrade (because of the PrusaConnect skipped line issue).
The issue happened with PrusaSlicer 2.8.0 and the latest 2.9.0, possibly earlier versions too.

Recently I did the upgrade from 2 tool heads to 5 tools heads and did a factory reset as suggested by the manual (which erased my statistics :( ), that did not change the behavior.
It does not seem to matter which tool is used, the error happened on Tool 1, 2 and 5 (probably 3 and 4 as well, just didn't test that yet).

crasbe, 2025-01-17:
I can add to the experience of the original author:

> The tools appear to be heating up based on the fan sounds but even after waiting a few minutes, nothing happens.

The tool does not heat up, all the LEDs stay green and the temperature remains at ambient temperature.

<br/>

> As a side note, after resetting, the printer refuses to establish a connection to Prusa Connect for quite a while.

This is not the case for me, the Prusa XL is connected via LAN in my case and connects to Prusa Connect instantly. This might be due to some of the improvements in the latest firmware and PrusaSlicer versions though.

<br/>

> Run a multi tool print (from Prusa Connect?)

It does not have to be a multi tool print in my experience. The last one I did had a filament change during the print though, so this might be related.

<br/>

> 1a) Spool join was used twice when this error occured - this may or may not be related

I can confirm that it did happen with spool join, I am not sure if it happened without.

<hr/>

Thinking about this, this might be the core issue: When a filament change happens during the print, the aforementioned state machine might not be uninitialized and therefore the state transitions are not happening.

crasbe, 2025-01-17:
1) I just tried to print a small cylinder with a filament change in the middle on Tool 1...

[image]

2) ... and a cylinder where the filament ran out (I cut the filament during the print) to provoke a filament change.

Both instances did _not_ cause the issue and the filament could be changed without any issues. Both prints were uploaded and started immediately with PrusaConnect.

3) Then I started the first print job again (without uploading it before) from PrusaConnect and the issue was not present.

4) Then I uploaded the cylinder to the printer sliced for Tool 2 and started it later with PrusaConnect and did a manual filament change (I forgot to update the change in PrusaSlicer and the M600 command was still for Tool 1 and therefore ignored) and the issue was not present.

5) Then I uploaded and started the cylinder on Tool 2 with a filament change in the middle and the issue was not present.

6) Then I uploaded and started the cylinder sliced for Tool 2 but the lower half being printed by Tool 1 using a height modifier and the issue was not present.

<hr/>

So perhaps it does have something to do with print time or volume as well.

Another thing to try might be to wait until the tool is fully cooled down. This was not the case for the tests above, but for the long print I mentioned before. I am not sure though if the tool was always cooled down when the issue occured.

kvtorp, 2025-10-16:
I am currently printing this model at 200%: [Giant Spider](https://www.printables.com/model/288796-giant-vase-spider-posable-halloween-print-in-vase-), in "many" copies for Halloween, and I am experiencing this a lot.
While my XL is multitool, this is executed as a single color print on one toolhead only (#2), in vase mode, typically 2-3 pieces on each bed, and thus one object at a time. Print times are typically exceeds 5 hours, the longest is more than 8, but nothing like the 12-20 hours reported earlier. So the prints are big, but thanks to vase mode they do not take forever.
The nature of the prints is so that Z is almost maxed out during the print, and the post-print bed drops the Z to its maximum. One of my plates is the pedipalps and mandibles, so a very low bed. I did not try a filament change after that, but one is printing right now, and I will try when it completes and put an update here, if it worked or not.

Prints are started either by queueing already loaded files from PrusaConnect and Set Ready on the printer, or directly from the printer.

If try to change filament mid-print, it works perfectly every time.
If I run out of filament mid-print, it works perfectly every time.
If I try to change the filament after the print, the printer is stuck at the initial please wait/hourglass screen every time. This is independent of whether I use Change filament in all tools or just Change filament and select the tool. The tool stays green, does not heat up, and the initial klonk that I usually hear when the XY steppers engage is not heard. Basically, the printer's physical state does not change in any perceptible way.
PrusaConnect connection does not seem to be impacted in any way.

The printer is an XL 5TH, printing off toolhead 2. Firmware is version 6.2.6+8948. The printer has previously been on firmware version 6.4.0-alpha. The printer is networked through Ethernet.

3d-gussner, 2025-10-16:
@kvtorp Thanks for the detailed comment.

kvtorp, 2025-10-16:
@3d-gussner You are so very welcome. I am currently printing a model that triggers this, so I can just as well gather all the data possible.
An extra detail: After finishing a short/low print, I moved the Z to 360 and asked to change filament. This works perfectly.
I am back to printing tall beds, and will try and ask for a filament change after each, to double check if it is really each and every time, or if I were just particularly unlucky