prusa3d/Prusa-Firmware-Buddy #3831: [ENHANCEMENT] <M500/Enhanced Experimental Menu>

Tlevendig, 2024-03-14:
**Printer type** - [MINI][MK4][XL]

The experimental menu allows you to save certain settings, but many important settings are still lacking. Think of PID values (preferably PID autotune), input shaping values (to allow users that have hardware modifications to update these), but also for example the filament loading settings on the mini (changing the length for example). 

All these things would have been possible with M500 to save the settings. Can these features be added to the experimental menu, or alternatively, can you enable M500 when in the experimental menu (so that there is a workaround to still use it).

Tupson444, 2024-06-11:
This is an important feature request. I understand there are reasons for not having M500 ability (I guess to prevent unwanted settings to remain active accidentally, or something), but it is not good not to have that option at all.

Maybe add "enable M500" in the experimental menu ?

aknezevi-work, 2026-02-04:
I also believe that this should be something which should be available to advanced users.
It should not be too hard to allow M500 while also having factory reset to reset the EEPROM, if that is for some reason not the case?