prusa3d/Prusa-Firmware-Buddy #3835: [BUG] Worse VFA with mk3.5

numo68, 2024-03-16:
**Printer type** - mk3.5

**Printer firmware version** - 5.2.1+13903

**Original or Custom firmware** - Original

**Optional upgrades** - Revo hotend, but the issue was present before

**USB drive or USB/Octoprint**
  USB drive

**Describe the bug**
  The vertical fine artifacts spaced 2mm, i.e. coming from the belt teeth, are very visible with the mk3.5. They were there but barely visible pre-upgrade.

  I have tuned the belts using the app, I have also replaced the pulleys on both X and Y motors. This did not help. Both idlers are original (smooth).

**How to reproduce**
  Create a box 51x49x10 mm (to be able to see which axis is which), slice with the 0.20 mm speed setting, Prusament PETG profile, no other changes, and print

**Expected behavior**
  VFA comparable with the mk3s+. Of course it could be just the result of a higher speed but in that case the profile should probably not allow external perimeters that fast. If there is however a change in the trinamic driver handling between the Rambo and xBuddy firmware, maybe it needs to be tuned, or perhaps the tuning made available for end-user experimenting.

**G-code**
  
[VFATest.zip containing the 3mf and bgcode](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/14623696/VFATest.zip)

**Crash dump file**
  N/A

**Video**
  Picture attached. Unfortunately no picture of the same object pre-upgrade. It is hard to get a good photo if the light source is near but the artifacts are in fact uniform and spaced exactly 2 mm. 
[image] [image]

numo68, 2024-03-16:
Update: slicing with the structural profile that is limiting the external perimeters to 45 mm/s is better and roughly matches the 3s+. Closing.

closed 2024-03-16 by numo68

DataPocalypse42, 2024-03-19:
Same with MK4, IS or not, the external perimeter should be 25mm/s to have an acceptable quality. 
It's not normal on a 1k printer, CS keep saying there nothing wrong, it's not acceptable