prusa3d/Prusa-Firmware-Buddy #3836: [QUESTION] Source code for 5.2.x

grzlus, 2024-03-16:
Hello,

I was able to install custom firmware, as my printer uses Mosquito Magnum with Bondtech Extruder and it works now pretty well. Now I see, that I could change most of the things through menu (Z height and steps), but I'd soon rebuilding the firmware, to support [MMU12x](https://github.com/cjbaar/prusa-mmu-12x).

So 2 things that I'd like to clarify:

1. Where should I look for a branch that contains latest firmware. `master` branch contains only 5.1, but in releases I see 5.2.
2. Why the do you put [it](https://github.com/prusa3d/Prusa-Firmware-Buddy/tree/master/lib/Prusa-Firmware-MMU) directly in repository? Why not sub repo? Will you archive the previous repo?

grzlus, 2024-03-29:
Now it's also about source code for 6.x

danopernis, 2024-03-29:
Source code for the 32-bit firmware is available at the respective git tag in this repository. Link is available at the releases page here on github if you are unfamiliar with how the git works. It will eventually be also merged into `master` branch. This will most likely happen after the final 6.0.0 release.

MMU firmware is imported to this repo to enable sharing some source code. Arguably this can be improved at some point, but at this point in time there are more pressing issues.

Hope that answers both of your questions.

grzlus, 2024-03-29:
Thanks @danopernis, I copied tags, as it's not included in forks by default. I'll keep it to use latest firmware version.

And for the MMU firmware, can I assume that the standalone repo is abandoned and fully use the one included in this repo? I made few changes there and I'd like to put them somewhere in my forks.

danopernis, 2024-03-29:
No, the standalone repository is not abandoned. IIRC it is pulled into this repository via `git subrepo`. Sorry for the inconvenience :-/ This has to do with supporting MMU also in the 8-bit firmware and as I said, might be improved later.

grzlus, 2024-03-29:
Ok, I'll keep using 2 repositories for code then.

Thank you for clarification 😄

closed 2024-03-29 by grzlus