prusa3d/Prusa-Firmware-Buddy #3839: [BUG] MK3.5 + MMU3 first layer calibration wait forever if you select from settings/hardware/steel sheet/type/

SNDR0, 2024-03-18:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3.5

**Printer firmware version** - 5.2.1

**Original or Custom firmware** - Original

**Optional upgrades** - MMU3

**Describe the bug**
  When you try to calibrate the first layer calibration via settings/hardware/steel sheet/smooth1, select filament type (pla), the printer start to heat only the extruder and wait forever the temperature. 
[image]

**How to reproduce**
  I preloaded to mmu3 only filament1. I tried using preheated for pla before, so also the heatbed heated up. nothing changed. it wait forever.
if I load to nozzle then first layer calibration works perfectly

**Expected behavior**
  it's little bug that not consider the mmu upgrade, than it wait that i insert manually the filament

**Video**
  Please attach a video. It usually helps to solve the problem.

bazzadark, 2024-03-29:
Bug also occurs if you try to use the calibration wizard direct as well !  Previous Mk3S/+ behaviour would always prompt for the MMU filament to use when running the 1st layer calibration.  Users will not see this problem when doing the initial Mk3.5 Conversion setup as the MMU3 cannot be configured/used till after the setup routine is finished (which includes 1st layer calibration check).

ljford7, 2024-03-30:
I can confirm this bug as well (MK3.5 with MMU3).

My workaround is to load to nozzle with the MMU3 and then do the first layer (it still isn't ideal, but works).

pandadoudou, 2024-03-31:
I have the same issue as well.

edooper, 2024-04-09:
I also have this issue, with latest FW (5.2.2)

danopernis, 2024-04-11:
Tracked internally as BFW-5139

bazzadark, 2024-06-11:
Dear Bot, Still awaiting update from internal issue BFW-5139.
thankyou