prusa3d/Prusa-Firmware-Buddy #3845: [BUG] <heating/cooling to 170C during the MBL delays the start of printing by minutes!>

Mercur13, 2024-03-21:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI, MK4, XL]
MK3.5
**Printer firmware version** - [e.g. 4.0.5, ...]
5.2.2 +13941
**Original or Custom firmware** - [Original/Custom]
Original
**Optional upgrades** - [e.g. Filament Runout Sensor]
no
**USB drive or USB/Octoprint**
  Please let us know if you print via USB flash drive or USB/Octoprint
Octoprint but also with Firstlayer-calibration (internally)
**Describe the bug**
  The additional step of heating/cooling to 170C during the MBL on the MK3.5 delays the start of printing by minutes! In my opinion its only relevant for MK4!
**How to reproduce**
  Everytime you Start the print.

**Expected behavior**
  A clear and concise description of what you expected to happen.
  Direkt heating nozzle/bed to intended temperature (behavior like the MK3S+ before)
**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.
not relevant
**Crash dump file**
  Please send the crash dump file to Prusa by emailing it to reports@prusa3d.com. Sharing this file is important and helps us investigate the bug.
not relevant
  Do not share the file publicly, as the crash dump contains a raw snapshot of the printer's memory and may include unencrypted sensitive information.

**Video**
  Please attach a video. It usually helps to solve the problem.

danopernis, 2024-03-21:
This doesn't seem to be a bug in firmware. Correct me if I am wrong, but firmware only executes gcodes. If this is a bug, it should be opened in the Prusa Slicer. Meanwhile, it can be worked around using custom gcodes in the slicer.

Mercur13, 2024-03-21:
But you are quick to close posts today!
The same behavior also occurs with the internal first shift test. So it's in the firmware.
Apparently this is an explicit MK4 routine, but it doesn't make any sense for the MK3.(5).
How should I introduce custom G-codes into the firmware at user level ?

woecj, 2024-04-08:
I observe the same behavior as Mercur13 reported. No solution for that? It's really annoying!
woec

Tupson444, 2024-04-08:
Just change the start code in PrusaSlicer (Printer Settings > Custom G-code > Start G-Code). Search for command `M109 R` or `M109 T0 R` (this sets the temperature and commands it to wait until reached). You can find more info about Gcode here: https://marlinfw.org/docs/gcode/M109.html

I guess if the printer profile for MK3.5 in PrusaSlicer is not optimal, you should open a request here https://github.com/prusa3d/PrusaSlicer-settings (if I'm not wrong).

rtyr, 2024-04-11:
@danopernis 
There is no such thing in the MK3.5 start g-code (feel free to check yourself). I suspect there is some unwanted FW behavior when running MK3 gcode on MK3.5 printer. See https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3896#issuecomment-2044713241 (that is MK3S start g-code, not MK3.5)

CZDanol, 2024-04-13:
Internal ticked number: BFW-5365

CZDanol, 2024-04-13:
The same issue here: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3896

Mercur13, 2024-04-13:
Closed because...? solved... or to ignore?

CZDanol, 2024-04-13:
> Closed because...? solved... or to ignore?

There's a dup issue that's open: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3896

While it would be formally better to have this one open and the other closed as dup, the state happens to be opposite here.

closed 2024-07-03 by danopernis