prusa3d/Prusa-Firmware-Buddy #3846: [ENHANCEMENT] <MK3.5: Poor Information during print.>

Mercur13, 2024-03-21:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - [MINI]
MK3.5
**Optional upgrades** - [e.g. Filament Runout Sensor]
no
**Is your suggested improvement related to an existing problem? Please describe.**
I'm always frustrated because I can't have needed Information during printing.
If I print over Octoprint, the Octoprint-Logo is not, what I,m expecting.

**Describe the expected functionality**
During printing, important information is missing from the print screen on the MK3.5 compared to the info display on my previous MK3S+:
- Print progress in % - only displayed during printing.
- Status bar (Prusa i3 MK3S ready / heating / model name.gcode, etc., IP address)
- Print speed (changeable)
- Estimation of remaining printing time
- Sheet steel type
- Position of the Z axis (can be displayed optionally, but this makes the font in the footer barely readable)

And please no annoying octopus anymore.

danopernis, 2024-03-21:
Duplicates #3825

Mercur13, 2024-03-21:
Sorry, #3825 reflects only to Octoprint. But the behavior is **generally** a problem !

danopernis, 2024-03-21:
Alright, I am going to do it the other way around. Close #3825 and keep this, at this is not really a bug, but more like a feature request/enhancement.

I can't give you any estimate on that, but patches are welcome.

Mercur13, 2024-03-21:
Thanks, I declared it already as Enhencement (but very urgent for MK3.5 users), because it means a deterioration compared to the original MK3S+ version.

timur-tabi, 2024-04-07:
I can't believe Prusa is shipping the MK3.5 with this stupid logo occupying half the screen and the only useful information being a tiny line on the bottom.

This report should be reclassified as a bug.  I upgraded my printer to a new version, and important functionality is missing.  By definition, that is a REGRESSION, and all regressions are bugs.

coaxial, 2024-04-14:
@timur-tabi it's only been over 4 years that the rest of us (Mini owners) have been waiting for this. Get yourself a comfy seat, looks like it's going to be a while longer if it isn't on any prusa roadmap at the moment.

redPanther, 2024-04-16:
You can workaround this issue like shown in my comment https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/189#issuecomment-873104604

I know this is not the solution, but you get information while we (endlessly) waiting that this issue get fixed. 

BTW "Sheet steel type" and "z height" is visible in the footer with latest firmware - at least on my mini. To be kind to the developers ... firmware 5.1.2 was a big octoprint update. I can now change the sheet type, load/unload filament while connected to ocoprint - this was not the case in prior versions and this was absolutely anoying.

flinke-flasche, 2024-04-17:
I use Repetier Server over USB and i'm missing the print informations like on my MK3s+.
Please implement this very important feature very fast.

Prusa-Support, 2024-04-26:
I seem to understand that some of the stock firmware features (especially "parametric" or even "modifiable" ones) have been internally tested in various cases of uses of serial connection and proved to be unstable or partially incompatible. Also, I seem to understand this would require rewriting large portions of the firmware from scratch, for third-party applications, which may even result in forking the original firmware.

We are willing to keep this goal in mind during future firmware developments, in accordance with the company's legitimate development plans. This said, we are unable to provide estimations for the implementation of these requests, and external contributions to our Open Source project, in this regard, are very welcome.

Michele Moramarco
Prusa Research

Mercur13, 2024-04-28:
Sorry Michele, you're unfortunately on the wrong track. It's not about support for third-party software (e.g. Octoprint) but about the performance features of the "Original Prusa firm/software" on printers of the MK3x generation (in my case the MK3S+). 
**We also want to see this information on the displays of the new MK4 generation!**

ttabi, 2024-04-28:
@Prusa-Support Michele, your response is very confusing. Are you sure you read the entire bug report?  All we're asking for is for the MK3.5 and later printers to show the same information on the LCD that the MK3s does today.  That's it.  This is not a request for new functionality.  This is a request to *restore* the *old* functionality that you removed.

hstag, 2024-05-05:
Any news on this issue?

Mercur13, 2024-05-05:
It seems like @Michele Moramarco is a bit shy about correcting her point of view, doesn't she???

Prusa-Support, 2024-06-07:
>It seems like @michele Moramarco is a bit shy about correcting her point of view, doesn't she???>

Probably you are right but one correction is due: Michele is an extremely popular male Italian name.
Not sure where you are going with that but let's not talk about me. Let's talk about the issue.

.
The use of the serial line is experimental, for advanced users, and is not considered a directly supported use in accordance with the manual. Octoprint and similar third-party applications are not directly supported by our company but we will not ignore yours and any other requests. Our developers will make considerations about this in the future, yet in observation of the company's legitimate development plans.

We are not denying your feature request and everyone is welcome to contribute to this Open Source project.
I'm sorry to see you uncomfortable with the answer but the previous replies are still valid.
https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3846#issuecomment-2013184464 - https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3846#issuecomment-2079805696

. 
Thank you @redPanther for the tip - https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3846#issuecomment-2058592759.

Michele Moramarco
Prusa Research

timur-tabi, 2024-06-07:
"The use of the serial line is experimental"

I'm sorry, but this isn't acceptable.  Prusa has been shipping products with serial connections for years, and they are a selling point.  You can no longer claim that they are "experimental".  It's a very bad look if Prusa cannot get 1960s technology working reliably on their hardware.

Mercur13, 2024-06-08:
Hello Michele, I'm sorry that I assigned the wrong gender to your first name. Sorry again!
But don't you think that your customers should be able to trust that the features that were available in the previous MK3S(+) printer generation should also be available in the next generation of printers from MK3.5 onwards.
Please look at this from the customer's perspective!
For us as customers, this is not an "extension" that will be installed at some point, but a bug that needs to be fixed!
PRUSA always emphasizes that Octoprint will continue to be supported, so why your negative attitude?

flinke-flasche, 2024-07-02:
hi,
is it so difficult to implement this feature?
It works on a 8-bit machine.
I love this feature on my MK3s+ and i want it on my MK3.5 as well.
Please implement it for tousands of users who use it.
(The most of them don't write here, but they want it)

By the way.
The big octoprint logo is a no-go. i use repetier server.

redPanther, 2024-07-02:
> . Thank you @redPanther for the tip - [#3846 (comment)](https://github.com/prusa3d/Prusa-Firmware-Buddy/

thx for recognizing me :-) 
as an intermediate solution wouldn't it be nice to have an option in settings where you can choose to have messages shown instead of the octo logo?
I mean something like this:
in setttings->user interface:
new item: Serial Printing [Logo]
you can choose between "Logo" and "Messages"

if "logo" is selected it is like now, if "messages" is selected, automatically open the "print messages" instead of displaying the logo.
This should be fairly easy to implement and test. Yes this is not the full featured stuff we know, but at least we get the info we need without additional actions.

Just to tell it again, I can only speak for the mini, because I have one, but there serial printing is flawlesly. All infomations are shown, ok not so pretty looking, but ok. 
Nevertheless it would be nice if we get (marlin) standard features back and it is a bit odd that so many people had to wait so long for that.
It wasn't a smart move from prusa to not address this prior mk4 release ...

hstag, 2024-07-04:
Octoprint is experimental? Really?

I was buying mk4 as an upgrade of the mk3s+ and I saw written nowhere, that octoprint is experimental, nor that octoprint is not supported anymore. This is a lie. Even on prusa websites it says mk4 has octoprint support.

Show me where this is written, that octoprint is experimental.

timur-tabi, 2024-07-04:
I think what he's trying to say is that the serial port connection on the MK3.5 is "experimental".  But that was a stupid thing to say regardless.

nealzip, 2024-09-01:
Do we have any update on this? At least if we could get ETA time on the display would be great.

timur-tabi, 2024-09-01:
I think https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3846#issuecomment-2154700645 made it pretty clear that Prusa has no intention of fixing this.

danopernis, 2024-09-02:
No ETA, still in our backlog with low priority.

TheNexusCore, 2025-08-25:
Still on backlog?
Just want to be sure after nearly a year since the last comment.

andrew64k, 2026-08-15:
I'll create a PR, for now I published it with binary firmware:
 https://github.com/andrew64k/Prusa-Firmware-Buddy/releases/tag/v6.5.7

I did make a portrait display version for the Mini, but there is no official 6.5.7 version of the code and I don't have a mini to test.... so I did not publish that binary version. @coaxial, do you want to test it?